opentrack/opentrack #935: Kinect Multi Tracker

Slion, 2019-04-26:
Drawing from experience gained while implementing Kinect Face Tracker and Easy Tracker I'm thinking to implement a Kinect Multi Tracker.

Such a tracker should be able to use multiple input from your Kinect device to provide the best face tracking experience.

sthalik, 2019-04-26:
What's the rationale here?

Slion, 2019-04-26:
I'm hoping to get better result than when using just the IR frame. I'm thinking of tracking the markers in IR, then refine their positioning in the colour buffer before solving the PnP problem using the colour camera. It could possibly provide better resolution or it could be a waste of time.

We could also use Kinect face tracking to help solve PnP too. The depth buffer could be of some use too, who knows.

sthalik, 2019-04-26:
Are you able to detect IR lighting on the color buffer?

Slion, 2019-04-27:
> Are you able to detect IR lighting on the color buffer?

Most certainly not with passive markers. But the Kinect SDK provides a coordinate mapper allowing you to match positions in the IR frame with positions in the Colour frame. So I figure if I get two corners of a bounding box in IR and translate them to Colour coordinates I get a more precise centre position.

Slion, 2019-05-19:
Postponed until further notice as Easy Tracker performance is very satisfactory.

closed 2019-05-19 by Slion