opentrack/opentrack #936: Best model for Easy Tracker OpenCV P3P solver

Slion, 2019-04-27:
With Easy Tracker and Kinect V2 hardware it looks like I'm getting a lot less noise when using my custom 3 vertices model instead of the default TrackerIR cap. Even though the coordinates of my custom model are certainly not as precise as the ones from the TrackerIR cap. It's so much better that the custom model signal could almost be used without filtering.

The picture below shows the TrackIR cap on the left and my custom model on the right.

[image]

Custom model specs:
[image]

Default TrackIR specs:
[image]

Slion, 2019-04-27:
I had a go at Elite Dangerous with another custom hat much like the one shown above and the experience is now much the same than with the TrackIR camera, just awesome 😄 Since the signal is a lot more stable I could lower the Accela setting to Smoothing 2.5 and Deadzone 0.5. I wonder if they could be lowered some more.

sthalik, 2019-04-28:
This isn't an issue when using a non-Kinect webcam. People using caps report excellent tracking and stability.

Slion, 2019-04-28:
> This isn't an issue when using a non-Kinect webcam. People using caps report excellent tracking and stability.

Of course this is something I observed on Kinect. Those guys do not use the same solver and they also are most likely using active markers and custom cap rather than the default TrackIR passive one.

closed 2019-05-19 by Slion