opentrack/opentrack #938: Separate plugin build

olukelo, 2019-04-29:
Is there any instructions how to build separate plugin without rebuilding whole app?

Slion, 2019-04-30:
With Visual Studio on Windows you only need to build the whole solution once and then rebuild only the one plug-in you are working on. 
See: https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows

sthalik, 2019-04-30:
On Windows you're going to run into ABI issues if trying to use a plugin with the existing binaries. Only if all relevant bits match (CRT used, Qt version, opentrack's header contents) you can be free of crashes.

So save yourself the trouble and ship an entire install directory rather than a single plugin.

closed 2019-05-10 by Slion