opentrack/opentrack #939: Camera intrinsic integration in OpenTrack API

Slion, 2019-04-30:
Until recently OpenTrack camera API was relying on the user providing camera specific variables such as FOV and frame dimensions.

This changed with the introduction of Easy Tracker using Microsoft Kinect V2 IR camera where camera intrinsic is provided by the camera implementation rather than its user thus changing the flow of information.

We would like to consolidate the camera API following those latest changes. 

See how the discussion started on https://github.com/opentrack/opentrack/pull/937

sthalik, 2019-04-30:
There are several issues I can remember at this moment:

- the focal length in pixels and the principal point can only be computed provided the FOV and frame size
- the FOV can change arbitrarily at any point in time due to the settings dialog
- `make_camera` returns `nullptr` in some cases, this causes crashes in multiple trackers
- `make_camera` should default to returning the first camera that exists if the named one doesn't exist
- regarding the distortion coefficents, I'll let you decide for the tracker you maintain
- there's also a crash in tracker/easy in `cv::cvtColor()` for regular webcams

Maybe the FOV value can be input to `struct info` then into `virtual void get_intrinsics(info&) = 0` or the intrinsics split into a separate struct.

sthalik, 2019-04-30:
Remembered few more bits:

- I see that now tracker/easy is in heavy development, though eventually it needs replacing `cv::Mat` with the typed variants
- Agreed on the float/double intrinsic types, feel free to push that without a pull request.

Slion, 2019-05-19:
> there's also a crash in tracker/easy in cv::cvtColor() for regular webcams

I've tried to keep colour buffer running but it won't be functional anyway. In fact the point extraction logic is currently assuming single channel. I'm now simply skipping point extraction if multiple channel detected. Check if that crash still occurs on your side.

Slion, 2019-05-19:
> tracker/easy is in heavy development

Not anymore. I'm not excluding doing major modifications in the future but for now I'm very happy with it. It looks like the heavy lifting is behind me already.

> needs replacing cv::Mat with the typed variants

I'm not familiar with those OpenCV finesse, could show me what you mean using an example?

sthalik, 2019-05-20:
> Not anymore. I'm not excluding doing major modifications in the future but for now I'm very happy with it. It looks like the heavy lifting is behind me already.

:+1: 

> I'm not familiar with those OpenCV finesse, could you show me what you mean using an example?

A 3x3 matrix with `double` element type is `cv::Matx33d`. Same size matrix with floats is `cv::Matx33d`. These are typedefs over `cv::Matx<typename element_type, int rows, int cols>`. This applies as long as the matrix size is statically known, i.e. not a video frame.

Then you don't need to call `cv::Mat::init` and the matrix becomes the same allocation as the struct containing it. The elements need to be initialized as normal, of course.

sthalik, 2019-05-20:
Just keep in mind that opencv doesn't validate matrix types. Even if the `matx` type is known, assertion failures are evident only at runtime when taking that code path.