opentrack/opentrack #944: Segfault on libevdev_uinput_get_fd()

waterlubber, 2019-05-04:
For whatever reason, I've been recently getting segfaults whenever trying to start tracking with the libevdev virtual joystick (as I can't use WINE because Elite is running in proton in a different prefix and wine version)
The program crashes with just a `libevdev error: -19`
This is the output with gdb, which hints at it being a problem when libevdev tries to get a file descriptor for /dev/uinput
https://pastebin.com/st2Gd5kk

I thought it was an rw issue, but I have permissions for `/dev/input`: 
```drwxr-xr-x   4 root root         580 May  4 06:46 input
crw-rw-rw-+  1 root root     10, 223 May  4 06:46 uinput
```
```
 waterlubber@gamingPC  ~  ls -hal /dev/input 
total 0
drwxr-xr-x   4 root root     580 May  4 06:46 .
drwxr-xr-x  21 root root    3.8K May  4 06:46 ..
drwxr-xr-x   2 root root     200 May  4 06:46 by-id
drwxr-xr-x   2 root root     220 May  4 06:46 by-path
crw-rw----   1 root input 13, 64 May  4 06:46 event0
crw-rw----   1 root input 13, 65 May  4 06:46 event1
crw-rw----   1 root input 13, 74 May  4 06:46 event10
crw-rw----   1 root input 13, 75 May  4 06:46 event11
crw-rw----   1 root input 13, 76 May  4 06:46 event12
crw-rw----   1 root input 13, 77 May  4 06:46 event13
crw-rw----   1 root input 13, 78 May  4 06:46 event14
crw-rw----   1 root input 13, 79 May  4 06:46 event15
crw-rw----   1 root input 13, 80 May  4 06:46 event16
crw-rw----   1 root input 13, 81 May  4 06:46 event17
crw-rw----   1 root input 13, 82 May  4 06:46 event18
crw-rw----   1 root input 13, 83 May  4 06:46 event19
crw-rw----+  1 root input 13, 66 May  4 06:46 event2
crw-rw----   1 root input 13, 84 May  4 06:46 event20
crw-rw----   1 root input 13, 85 May  4 06:46 event21
crw-rw----   1 root input 13, 67 May  4 06:46 event3
crw-rw----   1 root input 13, 68 May  4 06:46 event4
crw-rw----   1 root input 13, 69 May  4 06:46 event5
crw-rw----   1 root input 13, 70 May  4 06:46 event6
crw-rw----   1 root input 13, 71 May  4 06:46 event7
crw-rw----   1 root input 13, 72 May  4 06:46 event8
crw-rw----   1 root input 13, 73 May  4 06:46 event9
crw-rw-r--+  1 root input 13,  0 May  4 06:46 js0
crw-rw----   1 root input 13, 63 May  4 06:46 mice
crw-rw----   1 root input 13, 32 May  4 06:46 mouse0
```
```
 waterlubber@gamingPC  ~  groups
sys log users uucp input adm sudo waterlubber
```

The issue also occurs while running opentrack as root (haha yes I am big dumb) and with a freshly compiled version

I'm not entirely sure what's happening though

sthalik, 2019-05-04:
Please post a full stack trace.

sthalik, 2019-05-04:
I fixed error handling, errno as well as the failing call are displayed in a message box.

sthalik, 2019-05-04:
Try creating a symlink from `/dev/uinput` to `/dev/input/uinput`.

waterlubber, 2019-05-05:
Okay, so I tried a few things with a fresh compiled install.

- No more segfaults. Instead, I get proper error handling: 
```Error occurred while loading protocol libevdev joystick receiver

libevdev call 'libevdev_uinput_create_from_device(dev, LIBEVDEV_UINPUT_OPEN_MANAGED, &uidev)' failed with error 'No such device'```
Nice job

I tried the symlink and still didn't get it to work, however. Notable, running `sudo cat /dev/uinput` returns the same error (No such device.)

It appears that the `uinput` kernel module is not loaded for me. It may have something to do with my kernel version (5.0.10). I'll look into this in the coming hour and get back to you once I have something figured out.

waterlubber, 2019-05-05:
Figured it out. Apparently early today I had updated my kernel and forgot about it, meaning that all of the modules were in a new folder for the newer kernel but the old one was still running. Therefore, the uinput driver unloaded, and I was stuck wondering how my computer got so busted.

What a goofy way to figure that out

Thanks for the help, though, at least it won't segfault and leave me confused in the future.

closed 2019-05-05 by waterlubber