opentrack/opentrack #947: kinect connection for noob user

flodus, 2019-05-08:
Hello,
I downloaded opentrack 2.3.11 beta 3 for use with my kinect v2.
I have also installed the MS kinect v2 SDK.
But I don't understand how to make everything work.
I choose the input "kinect face 0.1" and the output "freetrack 2.0 improved".
When I press start I have my face in the frame preview video, but I move my head in all directions no axis changes, and the octopus does not move.
Are there any specific options?
I hope to make everything work.

Thank you for your work and for the community.

flodus, 2019-05-08:
thanks for the support.
I understand that slion was French so if he can answer me also in French it would help me a lot ;)

Slion, 2019-05-10:
> When I press start I have my face in the frame preview video

Do you get the frame around your face?
Try stepping back a bit from the Kinect and rise your hands until they are level with your head and move them around. That helps Kinect pick-up your skeleton and thus your face.
Also you should not need to install the Kinect SDK.

Slion, 2019-05-10:
Recule un peu, lève et bouge les bras à hauteur de tête. Cela aide le Kinect à identifier ton visage.

flodus, 2019-05-10:
super !!!
ça a marcher direct dès que je me suis reculé !!!
je commence à tester avec xrebirth...
je sens que ça va être le kiff !!!

----------

Great !!!
it worked directly as soon as I retreated !!!
I start testing with xrebirth ...
I feel like it's going to be the kiff !!!

flodus, 2019-05-10:
I will test already like that and I will look at Easy Tracker then!

Merci beaucoup
bests Regards

Slion, 2019-05-18:
@flodus Hope it's been working well for you.

I've since created some documentation about using Kinect with Open Track:
https://github.com/opentrack/opentrack/wiki/Using-Microsoft-Kinect-V2
Hope it helps.

closed 2019-05-18 by Slion