opentrack/opentrack #950: Document Kinect usage in Wiki

Slion, 2019-05-18:
Kinect usage needs to be documented in Wiki.

Also using this issue so that GitHub will host images used in Wiki without having to add them to OpenTrack repository. For some reason GitHub's Wiki does not support image upload like issues do. 
[image]

Slion, 2019-05-18:
Trash:
[image]

[image]

sthalik, 2019-05-18:
@Slion if you clone the wiki's `opentrack.wiki.git` repository then there's no problem with images.

Slion, 2019-05-18:
I see, I should have put them in the wiki repository, I guess. Though it worked well like that too.

Slion, 2019-05-18:
Wiki page is done:
https://github.com/opentrack/opentrack/wiki/Using-Microsoft-Kinect-V2

closed 2019-05-18 by Slion

StoryLabTAMU, 2019-07-26:
Hi Slion, I can't see Easy Tracker option appear in my input option, what should I do?

Slion, 2019-08-09:
Make sure you use one of the latest build.