opentrack/opentrack #953: Center key binding does not trigger action with specific keyboard

Slion, 2019-05-24:
Using one of my keyboards I can bind the Center action to F11 but the action won't trigger when using that specific keyboard. Other keyboards are working just fine.

Are we using different Windows API to record bindings and trigger actions?
Could you point me to places in the code where recording and trigger are happening?

It's very odd as I can see the F11 key press going through Windows HID stack like with any other keyboard.

sthalik, 2019-05-24:
We're using DirectInput. See the `dinput/keybinding-worker`. See if you can add a `qDebug` statement [here](https://github.com/opentrack/opentrack/blob/unstable/dinput/keybinding-worker.cpp#L189).

Slion, 2019-05-25:
It looks `pipeline::set_held_center` is being hit. I'm guess the issue is that, that funny keyboard of mines is sending the key up and down events too fast for this implementation to process it. Probably that bit is being reset to false before a center operation happens.

Slion, 2019-05-25:
That was indeed the issue. By making sure the key is being pressed at least 100ms it works. Not sure it's worth modifying Open Track implementation over that. It would be nice to have though.

sthalik, 2019-05-26:
Is this an autorepeat issue? Can you reproduce it in videogames, e.g. some WASD shooter?

Slion, 2019-05-29:
> Is this an autorepeat issue?

Not sure what you mean.

> Can you reproduce it in videogames, e.g. some WASD shooter?

It's not an issue in MWO.

The problem is that OpenTrack records the current state of keys, pressed or not pressed, rather than the fact that a key was pressed. Therefore if a key is pressed and released too fast for the pipeline to check key states between the press and release action then, as far as the pipeline is concern, that key was never pressed, even though it was.

sthalik, 2019-05-31:
> The problem is that OpenTrack records the current state of keys, pressed or not pressed, rather than the fact that a key was pressed. 

It's more comfortable for some people to center on key release. We can do both actually, introducing separate flags for the initial keypress and the keydown state. Performing center would clear the former flag. This way a spinlock can be avoided.

sthalik, 2019-05-31:
@Slion Would you like to give `logic/pipeline` a shot? Look for `enum bit_flags`. Set the new flag in `pipeline::set_held_center()` and clear in `pipeline::apply_center()`.

sthalik, 2019-05-31:
That is, `set_held_center()` should set it to true if the boolean argument is true, but do nothing if it's false.

sthalik, 2019-06-20:
@Slion please see if this fixes the issue.

closed 2020-01-27 by sthalik