opentrack/opentrack #955: Opentrack not recognizing camera input

Skilaufenmann, 2019-05-29:
I have been using Opentrack for some time and have not had any issues until now. The camera I am using is a PS3 eye with the free version of the CL drivers. The issue I am having is that, when I start Opentrack, it shows just a black screen on the picture in picture window and does not recognize any input from the camera. I know it is not the hardware at fault, as the camera still works through the CL-Test program. I sincerely appreciate any help that can be offered.

Slion, 2019-05-29:
Which version of OpenTrack are you using? When did the problem start, after updating OpenTrack?

Skilaufenmann, 2019-05-29:
Thank you for your response. At the time this issue came up, I was using 2.3.11 Beta 3. Though, I had been using this aforementioned version for about a week before I encountered this issue. Additionally, I uninstalled Opentrack after I encountered the issue and downgraded to 2.3.10, which did not end up remedying the issue. I noticed the problem earlier today. Additionally, I had not knowingly or purposefully made any changes to the program between the time that it was working and when it was not.

Slion, 2019-05-29:
Odd, maybe you could somehow reset the OpenTrack configuration to see if that helps.
Locate you profile directory and delete all INI files in there. Make a copy of those files before deleting them if ever you want to be able to get your configuration back to what it was.

Skilaufenmann, 2019-05-29:
That solved the issue. Thank you for your help

closed 2019-05-30 by Slion