opentrack/opentrack #956: Idea: a differenr deadzone

Iknowlongnamerightwhatwasithinking, 2019-05-30:
no idea how relevant this is to the actual program.... maybe it's a plugin feature.

So we have normal deadzone which just ignores input below a threshold. 
If we use DZ on yaw and pitch then we have a DZ that works along the whole axis arc. 
i.e 
If we have DZ on pitch then it is always their no matter where I look in yaw. 

In certain sims you only really want the DZ when looking directly forward. 
( nestle to gunsight etc...) 

SO.

With a bit of creative mathematics we could create a DZ that only works when you are 
inside a circle looking fwd. 
When you look out of this it is off. 

i can't code at all but.... I can see this is just a simple problem. 

Lets say the dz is 10 on both pitch and yaw.
if my pov is below 10 on yaw when I pitch up past 10 then the yaw DZ  will become 0.  
So when I yaw around ( above 10 pitch ) there is no yaw DZ.

I fly Battle of Stalingrad and at times I would like to focus on 
the sight BUT not be snapped in.  
If I need to look off bore a tiny bit I can just look out of the DZ.... and it's off outside of the DZ.

Just an idea! 

R.A.

Slion, 2019-05-30:
I pretty you can do that with mapping. However it's going to be a square or rectangle rather then a circle. Though you would not notice the difference anyway.

sthalik, 2019-05-31:
That's how the Accela filter does its deadzone, taking Euclidean distance from the transmitted position over rotation and position separately.

I can't possibly recommend doing deadzones using the mapping.

Iknowlongnamerightwhatwasithinking, 2019-06-03:
oh... I have not tried the Accela filter DZ. Thanks!

closed 2020-01-27 by sthalik