opentrack/opentrack #958: Z axis compensation

biffa66, 2019-06-04:
Maybe there is no fix to this and I cant have it both ways but here goes anyway...

I use Opentrack for combat flight sim head tracking  When I turn my head (yaw or pitch), my viewpoint suddenly translates on the Z axis unless I am already centred on Z over the seat and default position. Really annoying when tracking a target to side/rear and turning/looping. 

I can stop this annoying effect by ticking one of the disable options (roll axis I think!!) on the translation compensation screen (??).  This seems to effectively disable Z axis Compensation but what this then means is I have to move forward to increase my FOV when checking my 6 behind me. I just cant get used to this (moving forward to move back) in the heat of an engagement when looking to my rear. 
Is there a fix for this in Opentrack? Maybe a better combination of ticks in the 6 disable boxes (representing 6 DOF). Many thanks for any tips

Iknowlongnamerightwhatwasithinking, 2019-06-05:
Hello Biff.  ( it's me Plank ) 

Hey I am trying to understand your problem. 

A thing that I would do to try to solve your problem is to:
check the outputs. got to OPTIONS then the OUTPUT tab. 
Make sure yaw pitch x and y destination are set to yaw pitch x and y source. 
( this is confusing I know...) AND **disable** Roll  and Z.  

Now, make sure you are good with Yaw ( left right rotation) Pitch ( up down rotation) 
X ( left right translation ) and Y ( up down translation) . Get these working to your satisfaction. 

Now turn on Z ( fwd back translation ) Personally I never use it in Il-2 it never seems to 
work that well with my three point led target. ( gave up with it...) 

Essentially you need to focus on Yaw, pitch , x , y first. These are the biggies. 

The best thing to do in il-2 to get a good view over your shoulder is to set your
pilot position well forward, and fly with the canopy back ( yak, la, spitfire etc )

 I hope this helps! 

S! 

Planky.

sthalik, 2019-06-06:
Consider using the "dynamic neck" feature. It works regardless of relative translation state and whether Z is enabled or not. It's a copy of the same idea from DCS, where the eyes are offset forward from the axis of rotation.

biffa66, 2019-06-06:
Hi Plank and Stanislaw.  Thank you both very much for replying - it is very much appreciated.  I have now got this working the way I want it, although not exactly as you described.

(Stansilaw) When you say "Dynamic Neck" do you mean the "Neck Displacement" option under the Relative Translation tab?  I tried this but it didnt seem to make any difference to the response (in any way) that I could notice.  Perhaps I was doing something wrong.

(Plank).  I take your point but I really like to move forward and back in the cockpit (FOV in and out) when I want, but for it NOT to happen when my head pitches up, which was happening before.

The trick was enabling relative translation but disabling for Y and disabling for pitch, as image below

You can see the results in a short (pretty boring) video at https://www.youtube.com/watch?v=EjlfciC7_l4 and also a short in-game video at https://www.youtube.com/watch?v=UzX4F-uMLl4

Thanks again
Mark
[image]

biffa66, 2019-06-06:
Not my brainwork by the way.  The credit goes to Albertolg in his post under the thread _Relative translation off while held feature desapear. #712 on this forum

sthalik, 2019-06-07:
The neck displacement feature will work if you increase the forward position. It depends on your mapping.