opentrack/opentrack #959: Output Data

gomeslucasm, 2019-06-04:
what is the better way to catch the output data (X,Y,Z)  for use in other software, like puredata, matlab or python?

sthalik, 2019-06-06:
Either a [shared memory mapping](https://github.com/opentrack/opentrack/blob/unstable/freetrackclient/fttypes.h) or the UDP protocol. The former has less overhead.

billyjo2002, 2019-07-17:
Hi, I am something of a Luddite when it comes to new tricks and still an avid Visual Basic programmer (of sorts). Can you give me any simple explanation of how I can grab the X, Y, Z coordinates generated by Opentrack so I can use them in my VB application? I'd be grateful for any help.

sthalik, 2019-08-09:
Listen on a UDP socket and get the data.

closed 2020-01-27 by sthalik