opentrack/opentrack #960: Not returning to center

Geronimo553, 2019-06-05:
So I have been running build 2.3.9 for awhile now with a few mild issues. I noticed that .11 beta released a few months ago and updated. Now it seems that the tracking does not return to center if the trackir goes out of range. Instead of the view angle returning to center/default it instead gets stuck looking backwards or looking backwards upside down. 

Also vertical movement up and down has never worked correctly for me so I have to disable it. Any time I look left or right the camera will move up/down and forward/back. A total headache...

Anyway here are two images of the "not returning to center" issue of the latest build.

looking out of bounds
https://i.imgur.com/PJqMuao.png

not returned to center
https://i.imgur.com/iGZZBNI.png

sthalik, 2019-06-21:
Use the latest build from <<http://opentrack.misaki.pl>> and see if this is already fixed.

Also, set auto threshold (or pixel value) in the PT tracker so that point sizes correspond to what they actually are. Your point sizes are more of around 11 pixels and not twenty. Having them too big introduces false detections.

Geronimo553, 2019-08-10:
The problem with the point sizes is that anything under 18 pixel is not detected.

sthalik, 2019-09-22:
Can you adjust camera gain so the points are brighter than just few pixels above black? It's also going to help detection due to larger dynamic range.

closed 2020-01-27 by sthalik