opentrack/opentrack #961: 2.3.9 crashes when outputing virtual joystick Vjoy

Iknowlongnamerightwhatwasithinking, 2019-06-07:
When I press START OT crashes and win7 displays this :

..............................

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	opentrack Stick.exe
  Application Version:	0.0.0.0
  Application Timestamp:	599200f1
  Fault Module Name:	ntdll.dll
  Fault Module Version:	6.1.7601.24231
  Fault Module Timestamp:	5b6db230
  Exception Code:	c0000005
  Exception Offset:	0002e466
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789

........................

I can goad it into working by either starting OT and waiting a while before pressing START. 
Or set output to some other output and then start then stop then back to vjoy then wait then start. 

Is there a log file I can find to post here? ( I am not wise to such things..) 

Thanks! 

R.A.

sthalik, 2019-06-08:
Your best bet is to use a snapshot build at <<http://opentrack.misaki.pl>>. For now I can't make a stable release.

Iknowlongnamerightwhatwasithinking, 2019-06-09:
Hi. 

Those files seem corrupted. only 228kb each.  wont open. ( I use winrar ) 

Can I get 2.3.9  to generate a log file? to see what the crash is about?

Thanks! 

R.A.

sthalik, 2019-06-10:
> Those files seem corrupted. only 228kb each. wont open. ( I use winrar )

The archives  are eighty megs in size.

> Can I get 2.3.9 to generate a log file? to see what the crash is about?

I already fixed this crash and know what was causing it.

Iknowlongnamerightwhatwasithinking, 2019-06-11:
So how do I down load those files? 
I left click and save link as and it ends up saving a 225kb file. ?
I use firefox. 
If I right click on it I can open the archive and look inside it but...
that is less than optimal. 
Can you actually download the files? and how. 
Thanks! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-06-11:
oh god help me. It's the "download" button in the top right corner. ( ffs ) 

OK I good so far! 

Thank you! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-06-11:
Hey I don't think I made myself clear. 

I am trying to use 2.3.9 with the single point point tracker mod AND the vjoy output. 
the single point tracker won't work with any higher version. 
 
As it happens vjoy crashes in 2.3.11-beta-192 anyway... sigh. 

"Error occurred while loading protocol Joystick emulation -- vjoystick"

ok i will try to think around this... 

Thanks! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-06-11:
OK. 

I have a solution. 

I use two OT's.
I start OT1 on the head track camera. 
Then plug in camera two and start OT2. 
this is the stick tracking camera. 

Both use point tracker. 
OT1 outputs Freetrack 
OT2 outputs Vjoy. 

The trick is to use a three IR LED target for OT2. 

Rough testing works ok. 

The single point tracker mod that I used only works with 
OT 2.3.9 for some reason. If it was possible to get it to work with 
later versions... I would be happy as a clam. 

Thanks! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-06-12:
Actually.

Can you tell me why Vjoy crashes in 2.3.9 ???

I am pretty keen on keeping this version with the single point mod AND vjoy.

Why does it work "sometimes" then other times it just crashes all the time?

OR 

and this is probably out of the question... 

Can you work out how to make the single point mod work in the later versions 
of OT that don't crash Vjoy? 

single point mod.... 

For really really good 2dof  tracking 1point is the way to go. 

https://guuguu123456789.web.fc2.com/

Thanks! 

R.A.

sthalik, 2019-06-15:
> Can you tell me why Vjoy crashes in 2.3.9 ???

The vjoystick driver got updated and the old vjoystick library crashes with the new driver.

> Why does it work "sometimes" then other times it just crashes all the time?

Not all sources of bugs are deterministic. It's best if it crashes all the time to get fixed promptly, in counterintuitive terms.

> Can you work out how to make the single point mod work in the later versions of OT that don't crash Vjoy?

No, single point tracking is of poor quality and I won't endorse that. You can't get accurate yaw and pitch readings due to lack of data on the distance from the camera.

closed 2019-06-15 by sthalik