opentrack/opentrack #962: Opentrack doesn't recognize built in Bluetooth when connecting to Wiimotes

Silicon42, 2019-06-12:
Opentrack seems to be waiting for a USB Bluetooth dongle to be connected and shows me this image:
[image]
even though the Wiimote has been successfully paired with the PC using the latest version of the Toshiba Bluetooth stack (v9.10.34T) with the PC's built in Bluetooth adapter. I can tell that the Wiimote is connected and working properly because the same setup works in FreeTrack, however I was hoping to make use of Opentrack's better compatibility and current support. I tried looking at the code, but the lack of any explanatory comments makes it really hard to tell what is going on so I can figure out what is wrong.

If I had to hazard a guess though, it's probably not noticing the built in Bluetooth adapter, most likely due to it not being a USB device, in wii_camera.cpp in the wii_camera_status WIICamera::pair() function.

I'm running this on a Dell Inspiron 5548 running 64 bit Windows 7 with Opentrack 2.3.11 BETA3 if that helps any.

sthalik, 2019-06-15:
Please see if the issue is still present in the newest developer build on <<http://opentrack.misaki.pl>>. There was a pairing issue that only got fixed in March.

Silicon42, 2019-06-26:
Yep, that seems to have fixed it.

closed 2019-06-26 by Silicon42