opentrack/opentrack #965: Can opentrack crash rise of flight?

Iknowlongnamerightwhatwasithinking, 2019-06-24:
Rise of flight is constantly crashing for me. 

MSVCR90.dll faults and RoF crashes to desktop. 
( it happens after a sortie and as I press the "fly now" respawn button in multipalyer ) 

I have been asked by RoF support if my tracker is causing the issue. 
at this point I have not tested RoF with opentrack and without opentrack. 
No A/B test yet. Will do ASAP. 

There have been a lot of strange RoF crashes and some users will be using Opentrack. 

no one seems to have a clue as to what is going on. 
I have posted a thread on the il-2 sturmovik forum it's easy to find. 
It's mostly ranty whingy why me type stuff. Nothing I have tried works. etc. 

Help?

R.A.

MathijsG, 2019-06-25:
The very first thing I should try in deducting this, would be to try to run the program without opentrack, then you easily can eliminate the question whether or not Opentrack is related. I don't have a single problem with opentrack in combination with Rise of Flight though.

Iknowlongnamerightwhatwasithinking, 2019-06-25:
Hello. 

Started doing that yesterday. It was a slow process and I found it hard to keep track. 
The crashing is not regular. It's RANDOM. but always when I press Fly now. 
I can respawn many times with no issue then crash. 

Obviously this makes testing a super grind. 
I cannot make the crash happen at will. Vexing it is. 

I was ASKED by the 1CGS if my tracker maybe the problem. and here I am . 

Believe me I am trying all avenues but heck, I have NO IDEA what MSVCR90.dll is 
AND i have no idea why it faults, who is using it??? RoF ? Opentrack? Vjoy ? PPjoy ?   
MSI afterburner? AMD graphics drivers? 

It's all magic to me... sigh... should have studied programming in school. 

Thanks though. : - ) 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-06-26:
Just crashed Rise of flight without Opentrack running. 
This is good news for opentrack and I can stop trying to learn how to POV hat switch look... 
Thanks! 

R.A.

MathijsG, 2019-06-27:
Okay nice to hear it's not a bug in opentrack. You can close the issue now :).