opentrack/opentrack #968: tracker/easy & kinect wait for frame

sthalik, 2019-07-05:
@SLion,
There's an issue in tracker/easy where you specifically wait for 1000/Hz milliseconds for the next frame to arrive. Does this imply that the Kinect source returns the old frame rather than block until the next one arrives? That behavior wouldn't be in line with opentrack's video input API.

There are some issues where `iTicker` from tracker/easy complains about being destroyed from the wrong thread. Can you check this on your end?

Ideally you'd subclass `QThread' and run that kind of blocking code from overridden `QThread::run()'. This timer usage smells dubious.

I'd also like to get rid of iostreams but that's for another day.

Slion, 2019-07-05:
> There's an issue in tracker/easy where you specifically wait for 1000/Hz milliseconds for the next frame to arrive.

Where is that in the code?

> Does this imply that the Kinect source returns the old frame rather than block until the next one arrives?

I guess I'm just asking the Kinect API to provide me the current frame. I'm not sure what that API does but I doubt it blocks and waits for the next frame.

>  Can you check this on your end?

I guess I could whenever I get a chance.

> Ideally you'd subclass QThread and run that kind of blocking code from overridden QThread::run(). This timer usage smells dubious.

It was using `QThread` and spinning much like tracker-pt. Spinning was first quick fixed 1fb6122dce1a5bc117e925006d87828a52cfecb6 then properly addressed using a timer rather than a raw thread 088344f1bde14a24f90852316d01e6f1826da034.

> I'd also like to get rid of iostreams but that's for another day.

Ok, and another issue please otherwise it gets confusing 😃

sthalik, 2019-07-06:
>> There's an issue in tracker/easy where you specifically wait for 1000/Hz milliseconds for the next frame to arrive.

> Where is that in the code?

https://github.com/opentrack/opentrack/blob/unstable/tracker-easy/tracker-easy.cpp#L696

> I guess I'm just asking the Kinect API to provide me the current frame. I'm not sure what that API does but I doubt it blocks and waits for the next frame.

Please fix this. I ran into some deadlock with tracker/easy and can't fix this without having the Kinect camera provider work correctly.

> It was using QThread and spinning much like tracker-pt. Spinning was first quick fixed 1fb6122 then properly addressed using a timer rather than a raw thread 088344f.

With other camera input modules, tracker/pt spends most time sleeping.

Slion, 2019-07-08:
> I ran into some deadlock with tracker/easy and can't fix this without having the Kinect camera provider work correctly.

With which camera did you try tracker/easy? Can you reproduce the deadlock? How?
I don't have access to other cameras ATM so it's going to be hard for me to reproduce if it does not occur with Kinect IR cam. I did fix a deadlock issue there 48b80fb28b9e6e44267c9d1edb8a9968ff81fea5.

> With other camera input modules, tracker/pt spends most time sleeping.

I see. That would indeed be the case if the camera object sleeps until the next frame is available.
There could be a way to change the camera implementation, probably not that trivial though.

sthalik, 2019-07-09:
> With which camera did you try tracker/easy? Can you reproduce the deadlock? How?

The camera is the `video-opencv` provider. The reproducing case is enabling the debug window, then stopping the tracker. The main thread is in the destructor at `QThread::wait()`. The worker thread is at `cv::waitKey(1)`. This might be a threading issue with Qt signals getting fired from the wrong thread.

sthalik, 2019-07-09:
> I see. That would indeed be the case if the camera object sleeps until the next frame is available.
There could be a way to change the camera implementation, probably not that trivial though.

Don't you get any timecode metadata, or a Windows event to wait for the next frame?

sthalik, 2019-08-09:
@Slion do you mind if there's a sleep statement? Furthermore, we can adjust for inaccuracy in sleep time using the same scheme as `logic/pipeline`. Having it spin is way bad and affects all trackers.

Slion, 2019-08-09:
Yes there is an event you can get when a new frame is available.
Sleep sucks, timers usually do that for you.
Please don't commit delicate changes to master unless it was tested with Kinect.

sthalik, 2019-09-22:
> Yes there is an event you can get when a new frame is available.

Can you do this within 2-3 months please?

> Please don't commit delicate changes to master unless it was tested with Kinect.

Did I do that so far? I'm trying to go with pull requests for Kinect changes.

Slion, 2019-10-13:
> Can you do this within 2-3 months please?

Unlikely, I've been mostly away from my desk and my Kinect since July 2019 and it will remain like that for at least another month I'm afraid.

sthalik, 2019-10-22:
Alright. Shall the release wait or is it good enough? And what if I added a sleep statement? :smiling_imp:

Slion, 2019-10-22:
It's definitly production quality as it is and a proper release is long overdue. There is no spinning, CPU usage is optimal and a sleep statement is not needed since a timer is used instead.

sthalik, 2020-01-27:
> There is no spinning, CPU usage is optimal 

The Kinect plugin doesn't support the video API properly. The requirement is never to return stale idempotent frames. I'd be merging #1026 unless a better solution is found.

closed 2020-01-27 by sthalik