opentrack/opentrack #969: Support multiple

altimumdelta, 2019-07-13:
Wii Remotes have a kind of narrow FOV, I'm building my setup as we speak but I'm not sure how well is this going to perform on a 32inch 1440p screen inside DCS World where I would be trying to lean down and to the back around the cockpit, which is the main point, to be able to get up close to the gauges, and it looks like I won't be able to do that with Wii Tracking, but we'll see, perhaps some good settings and adjustments are possible so I could have Wii Remote placed up in an angle looking down so it can track when I'm looking closer at the screen, or use some high sensetivity for the X(forward-back) axis

HugoSandell, 2019-07-13:
I agree. In fact, I was considering opening an issue to see you've beaten me by about an hour.  
I've been trying to do it myself, but it's a headache to set up on my computer.

sthalik, 2020-01-27:
No way. This is too specialized to ever be viable.

closed 2020-01-27 by sthalik