opentrack/opentrack #970: Voice controlled tracking?

altimumdelta, 2019-07-13:
So I had an idea it could be fun to do it as simple as making certain frequency with voice and it would move you forward/up/down etc, it would definitely be wonky as you can only produce one sound at a time with mouth but it could at least be just one more way this great program can it, if nothing else just for the challenge and legacy of it, has anyone tried anything similar or some kind of translation program exists for UDP?

You would produce sounds basically like, aaaaa ,iiiii ,ooooo, uuuu, iiiiiii, aaaaa, oooo, iiiii aaaaa oooo _iiiiiao .... fiiaaoo fiiigaro fiiigaro figaro figaro ... ehm_ which would correspond to some frequency pattern, you would have to ofcourse sample your own voice but only like what minimum of 6 voices for 3D, and additional leaning etc but it gets more comlicated to remember ofcourse and then to use in practice, so it wouldn't be any kind of serious voice recognition at all, just a monotone, it could also be by volume in addition.

Would need some program that converts the sounds to the UDP stuff perhaps, I heard something about voice mentioned in the issue referenced below.
https://github.com/opentrack/opentrack/issues/576

sthalik, 2019-07-13:
Are you able to at least whistle out the DTMF codes?

altimumdelta, 2019-07-14:
Heh, I've updated the OP to mention that IMO *all* the software would need to do with the voice data is just calculate probability weight of that voice signal "matching" the user-set pattern/frequency (I say pattern just cause I don't have a better word now, it's not a pattern it's just one tone but you know there's tiny variations even in that from person to person, to notify it's something user-set, but I think the term "tone" should be more appropriate going forward). And it doesn't need to match, just how close it is, then whichever is the highest at any given point it'll apply the command tied to that frequency/tone.

I'm kinda new to head tracking and what I'm doing right now is emulating TrackIR so maybe there's better use of this in other kinds of "tracking".

closed 2020-01-27 by sthalik