opentrack/opentrack #971: Kinect-easy tracker: Please explain vertices setup?

wostg, 2019-07-13:
Hi, I'm a long time pointtracker user, now with a Kinect V2. I have had success with the facetracker, but am spending this Saturday night, like many, with an array of balls in my face.
May I ask for some support on the Easy Tracker vertices setup? I am using the 30 April version of OT with Easy Tracker and am not sure how to configure properly. 
I believe I have the markers set up close to the default cap setup 60x100x40/40.
Thank you for all the excellent work.

wostg, 2019-07-15:
eh seems fine

closed 2019-07-15 by wostg

Slion, 2019-07-19:
Glad you got it working.

See wiki: https://github.com/opentrack/opentrack/wiki/Using-Microsoft-Kinect-V2

Though to be fair I should really rewrite the Easy Tracker page:
https://github.com/opentrack/opentrack/wiki/Easy-Tracker

wostg, 2019-07-19:
Thanks for reply :) I am using kinect facetracking in daylight hours and easy tracker when it gets dark now. The vertices were logical enough once tested, but I don't think I've seen that layout within opentrack before so took me a minute.

May I ask what the benefit of 4/5/6 markers would realistically be?

Also I don't know if this is any use but thought I'd share the ritual I need to get easy tracker working: (possibly because I have MyDocs in Onedrive?) I need to start kinect up in the latest main Opentrack release (no easy tracker option) then close that down and load up the test build, run kinect, and then easy tracker.

Slion, 2019-07-24:
> May I ask what the benefit of 4/5/6 markers would realistically be?

Only three four or five vertices are supported. Just stick to three vertices unless you know what you are doing.

It was developed to try out various algorithms OpenCV is providing to solve the PNP problem. However testing with four and five vertices proved rather disappointing.

If I recall well four is notably bad since it often provides the wrong solution.

Five was alright but you get a lot more chances of vertices merging with one another thus breaking the solver.

wostg, 2019-07-24:
That's interesting to know, thank you!

StoryLabTAMU, 2019-07-26:
Thank you Slion:) When I went to Easy Tracker page, I don't know how to make a self-custom hat, can you send me a link or add some more explanation in there?

wostg, 2019-07-26:
Hey here is my hat, using the above numbers. It is made of
1 x black cap
3 x bouncy rubber balls covered with reflective cyclists tape
velcro dots to hold them on
[image]

StoryLabTAMU, 2019-07-28:
> Hey here is my hat, using the above numbers. It is made of
> 1 x black cap
> 3 x bouncy rubber balls covered with reflective cyclists tape
> velcro dots to hold them on
> [image]

Thank you wostg:)

sthalik, 2019-08-09:
> If I recall well four is notably bad since it often provides the wrong solution.

It's alright if it's not planar.

JimmyRaoUF, 2019-10-09:
> Hey here is my hat, using the above numbers. It is made of
> 1 x black cap
> 3 x bouncy rubber balls covered with reflective cyclists tape
> velcro dots to hold them on
> [image]

@wostg Hi,
can you share a screenshot of your settings?
I am having a problem with this:
https://github.com/opentrack/opentrack/issues/861#issuecomment-539760256

Can you switch between the IR source?

Thanks!

wostg, 2019-10-12:
Hi I think the below is everything. I think the numbers are probably reversed so inverted in the output settings. I put away the ps3eye once I got this working so not sure about switching sources!

[image]

Slion, 2019-10-13:
I wonder if I should fix that reverse problem in the code somehow.

Slion, 2019-10-13:
@wostg Your screenshots are showing Kinect V2 Video Sensor is selected. I seriously doubt that's working. It should be Kinect V2 IR Sensor.

wostg, 2019-10-13:
[image]
That's me now. I'm using whatever ms drivers and the regedit needed to use the kinect as a webcam if that helps.
To get it working I load up 2.3.11-beta3, run face 0.1, close that off, load up the beta above, run face 0.1 and then easy tracker 1.0. Assumed that was just the cost of running test versions.

Slion, 2019-10-13:
We ought to do a proper release.

JimmyRaoUF, 2019-10-13:
> To get it working I load up 2.3.11-beta3, run face 0.1, close that off, load up the beta above, run face 0.1 and then easy tracker 1.0. Assumed that was just the cost of running test versions.

@wostg Wow, I followed your instruction and it worked this time!
Big thanks to you!

@Slion Thank you for this wonderful work!

sthalik, 2019-10-22:
@Slion regarding the release, see #778. Can you fix waiting for the camera frame in the Kinect camera provider before that?