opentrack/opentrack #972: Can't save settings

RadioMan71, 2019-07-14:
Hello,

I got this issue with the program, at first I thought it was only the child project, Trackhat opentrack, but now that I installed the latest 2.3.11 release of the main program I am having the same issues.
I know there has been reports about this in the past, but those bugs have all been solved.

I can create a new empty or copied config file, so it doesn't appear to be some sort of read only conflict as I am able to write those files to disk, the only problem is that they stay empty, no matter what I do they remain 0 KB.
I used this program in the past around January of this year and the settings did save back then, but unfortunately I had to uninstall this because I thought it was conflicting with my tobii eyetracking software, but in the end that was facetracknoir.
Anyway, ever since I reinstalled this this problem occurs.
I also can't be sure about whether or not there used to be a way to manually save the profile, probably in the same place as you had to create a new profile, but that's not here also, so there is no way in testing a manual save unless there is, I would gladly try that out.
Hope you can help me with this and if you need some more information I try to give that to you too.

Len62, 2019-07-14:
This may not matter, however make sure you are running OT at the admin level.  Also, in the main window, click on the hammer next to the output section, and in the select interface option try setting it to enable both, that was an issue that was fixed sometime back, but you never know. Can you load the default ini file?

RadioMan71, 2019-07-15:
I am running it as an administrator, but this reminded me that I didn't do the install as an admin, so I deleted the entire settings directory and let the installer run as admin, it created the settings directory again, opened up opentrack, but it didn't want to open any profile, so I checked the directory it created and found that the extensions were all wrong, maybe this would help.
I did a dir >dir.txt in a dos window so I copy the results below.
[dir.txt](https://github.com/opentrack/opentrack/files/3390911/dir.txt)

sthalik, 2019-07-15:
Set your documents directory outside OneDrive.

RadioMan71, 2019-07-15:
I'm working on it, it's pretty slow because there are a lot of small files inside the folder, just hope it works, as it did work from within onedrive before.  Anyway, I couldn't just move the location, I had to copy the files to the new location, delete the original files and then move the folder location to the new location and this is taking a lot of time, since there are so many small files and that is severely hurting my copy speed.

RadioMan71, 2019-07-15:
Well that did the trick, only problem now is that my automatic backup to onedrive isn't working anymore, but maybe I can exclude the opentrack folder from the backup, but I got to look into that.  It took way longer than expected because well I got over 1 TB space on my onedrive and the documents folder alone was almost 300 GB in size and moving that to the onedrive recycle bin. But for future reference, this wasn't necessary at all, because it still didn't make me move the documents folder location, but turning off onedrive to backup the documents did work.

closed 2019-07-15 by RadioMan71

sthalik, 2019-07-15:
OneDrive is opening the files before opentrack is done writing them. opentrack then can't move the files into their expected extensions (`.ini`) and that's what you get.

Len62, 2019-07-15:
Something new learned!