opentrack/opentrack #974: PointTracker - pose interpretations and preffered one

gb2111, 2019-07-23:
Hi,
I am using Cap and when I go to the edge my pose goes away and octopusy turns around. i noted that it is related to a cross on my led not being on the top led any more but instead going to one of bottom ones. I understand that PointTracker thinks that my head is upside down.
I think there can be multiple interpretations of pose from 3 points. But the pose here does not make any sense. Is there way to resolve this?
https://gyazo.com/17f28cb44410d14e3e6a976a6983da2b

Thanks.

Len62, 2019-07-23:
Looks like that led gets lost when the pix falls below 7. Try reducing the minimum size for the point extraction.

gb2111, 2019-07-23:
there was no loss in tracking. i tested it many times. also in posted gif there is no visible loss of trackig

Len62, 2019-07-23:
Sorry, I misread the GIF. Have you run through the calibration procedure? Typically, the yellow cross is not directly over an led.

gb2111, 2019-07-24:
i have not. have not seen that in tutorials. do you have any that cover this part?

Len62, 2019-07-24:
Take a look at this video: https://www.youtube.com/watch?v=ZDz-bholoMo

Len62, 2019-07-25:
@gb2111 Have you managed to get this sorted out?

gb2111, 2019-07-26:
yes. the same thing happens. it seems this is just offset.
basically when I lose one of bottom leds and when it is back. opentrack thinks it is upper one and flips the octopusy. but it happens only when tracker is rotated almost 90 dagree.
you can see it very good here.
https://gyazo.com/b252f9256f6c867a11727f4d69b30149
then i need to cover all and once uncover they are fine.

i tried same with other model and it seems problem is not there.
i realize this is angle where you normally don't rotate when watching monitor :)

Len62, 2019-07-26:
You could try and reduce the max point extraction to 10 or so. That can help with the overlap of the LED's. What resolution do you have set?

sthalik, 2019-08-09:
@Len62 thanks for the assistance! I'm a bit unavailable at the moment.

sthalik, 2019-08-09:
@gb2111 what's the raw pose when you go outside the screen and back again? I need the raw values.

closed 2020-01-27 by sthalik