opentrack/opentrack #976: initial model centre

dylanmenzies, 2019-07-31:
I have an application where I would like the initial position centre to be fixed, eg 0 at the camera position. Currently the centre resets to the first captured reading. I imagine this requires some code modifications - can anyone give some pointers? Thanks.

sthalik, 2019-08-09:
You can disable initial center in the options dialog.

dylanmenzies, 2019-08-14:
@sthalik Thanks. When I disable 'centre at startup'  the z value is fixed at 75, so I can't track position. Can this be fixed?

sthalik, 2019-09-22:
That's the uncentered position.

sthalik, 2019-09-22:
That's the uncentered position. What is it that you're trying to accomplish exactly? Why is it that you can't use centering?

dylanmenzies, 2019-09-22:
I want to have the position relative to the cam, this is useful information in my application. Seems strange just the z value is fixed.

sthalik, 2019-10-12:
For the latest beta, raw values for the freetrack protocol are now filled. Get it from <<http://opentrack.misaki.pl>>.

dylanmenzies, 2019-10-12:
Thanks!
Dylan

sthalik, 2019-10-21:
@dylanmenzies can you confirm whether this works? Thanks!

Slion, 2019-10-22:
I had a similar request not so long ago, see #917. I worked around it but I'm still convinced that's the right thing to do.

dylanmenzies, 2019-10-22:
I just tried the beta. The z value is no linger flixed, but it reaches a value of 75 when the target is short distance from the cam then stays at that value as the target moves further away.

sthalik, 2019-10-24:
The raw values are only exposed through the in-memory mapping. See https://github.com/opentrack/opentrack/blob/unstable/freetrackclient/freetrackclient.c

closed 2020-01-27 by sthalik