opentrack/opentrack #977: kinect face 0.1 low fps high latency

denut, 2019-08-01:
the face tracker have low fps an high latency 

is there somthing to do  ?

denut, 2019-08-01:
tested with w10 camera no latency high fps

Slion, 2019-08-06:
How did you check that?
I guess the implementation could be improved.

denut, 2019-08-06:
the visualisation si not so fast in the 2 software by moving hands in open trach there is 0.5s delay and no delay in windows camera

Slion, 2019-08-12:
For best performance you could use your Kinect with Easy Tracker.
See:
https://github.com/opentrack/opentrack/wiki/Using-Microsoft-Kinect-V2

Slion, 2019-08-29:
I just tried Kinect Face Tracker today and there was a massive lag for some reason. After a reboot it was fine again. I reckon re-plugging your Kinect should also do the trick. I did not do any precise measurement but it did not feel laggy then.

closed 2019-08-29 by Slion

dijicaek, 2020-05-14:
Even though this is closed, I just wanted to chime in here in case anyone else comes across this issue. I had the low framerate problem with the Kinect V2 both with the Kinect Face 0.1 tracker and Easy Tracker 1.0. A reboot of the PC fixes it as per @Slion's suggestion. Disconnecting the Kinect, disabling the Kinect in device manager, and restarting Opentrack **don't** fix the issue. So yeah it's probably something outside of Opentrack.

searching46dof, 2026-09-21:
I had this symptom where reboots did not fix the issue.

Opentrack KinectFace video preview showed either of two symptoms 
- work for 10secs, then freeze for 5 secs, and repeat
- video preview doesn't freeze but face tracking rectangle freezes

During the freeze, the Kinect's lights turn off but it's power supply light stays on.

It was triggered by some windows update and the solution to disable Kinect Audio Enhancements worked for me.
https://brekel.com/kinect-disconnect-fix/
1. Open Windows Settings and go to System > Sound. Scroll to the bottom, to Advanced, and click More sound settings.
2. In the panel that opens, select the Recording tab.
3. Right-click Microphone Array - Xbox NUI Sensor and choose Properties.
4. On the Advanced tab, untick Enable audio enhancements under Signal Enhancements, and click OK.
5. Repeat for every Kinect entry in the Recording tab, not only the active one.