opentrack/opentrack #978: WMR controllers

point1024, 2019-08-12:
Will it be possible to track WMR controllers using point tracker and Bluetooth. Without using a mixed reality portal. Without a headset.

sthalik, 2019-09-22:
No idea whether you'd be able to mount a clip on that particular VR headset.

point1024, 2019-09-22:
no need to make a clip. controllers themselves have an LED circuit. the question is in tracking it.

sthalik, 2019-09-22:
I'd need to know the LED layout.

point1024, 2019-09-22:
https://docs.microsoft.com/en-us/windows/mixed-reality/motion-controllers.

if necessary I can measure the location directly on the controllers

point1024, 2019-09-22:
14 LED on the inside and 16 LED on the outside.

sthalik, 2019-09-22:
All trackers require the same point count. That's the biggest obstacle since it's non-trivial to implement. You may find the Aruco tracker interesting though. There's a page dedicated on the wiki.

point1024, 2019-09-22:
The device itself monitors the position of the controllers well, but as soon as they leave the field of view of the cameras, they cease to be normally tracked. and tracking only happens with the help of gyroscopes. It was thought to use an additional camera for tracking. but I feel the implementation will not be easy.

point1024, 2019-11-17:
http://freepascal.ru/forum/viewtopic.php?t=25002&view=unread#unread
Человек ковыряет в wmr уже есть доступ к данным wmr контроллеров.

sthalik, 2020-01-27:
Send a pull request with working code.

closed 2020-01-27 by sthalik