opentrack/opentrack #979: Exception when trying to start debug on Windows from VS 2019

Slion, 2019-08-13:
Exception thrown: read access violation.
```
 	opentrack-migration.dll!std::vector<std::function<std::shared_ptr<migrations::migration> __cdecl(void)>,std::allocator<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> > >::_Orphan_range(std::function<std::shared_ptr<migrations::migration> __cdecl(void)> * _First, std::function<std::shared_ptr<migrations::migration> __cdecl(void)> * _Last) Line 1687	C++
 	opentrack-migration.dll!std::vector<std::function<std::shared_ptr<migrations::migration> __cdecl(void)>,std::allocator<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> > >::_Emplace_back_with_unused_capacity<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> const &>(const std::function<std::shared_ptr<migrations::migration> __cdecl(void)> & <_Val_0>) Line 646	C++
 	opentrack-migration.dll!std::vector<std::function<std::shared_ptr<migrations::migration> __cdecl(void)>,std::allocator<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> > >::emplace_back<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> const &>(const std::function<std::shared_ptr<migrations::migration> __cdecl(void)> & <_Val_0>) Line 662	C++
 	opentrack-migration.dll!std::vector<std::function<std::shared_ptr<migrations::migration> __cdecl(void)>,std::allocator<std::function<std::shared_ptr<migrations::migration> __cdecl(void)> > >::push_back(const std::function<std::shared_ptr<migrations::migration> __cdecl(void)> & _Val) Line 675	C++
>	opentrack-migration.dll!migrations::detail::migrator::add_migration_thunk(std::function<std::shared_ptr<migrations::migration> __cdecl(void)> & thunk) Line 86	C++
 	opentrack-migration.dll!migrations::detail::registrator<max_pitch_output>::registrator<max_pitch_output>() Line 56	C++
 	opentrack-migration.dll!`dynamic initializer for 'init_0''() Line 70	C++
 	[External Code]	
```

sthalik, 2019-08-16:
Does that occur with my dropbox binaries?

Slion, 2019-08-20:
I just tried 20190430_01 and it starts. However EasyTracker is broken in that build.

Slion, 2019-08-28:
The above issue is apparently a regression introduced in VS2019 to do with _ITERATOR_DEBUG_LEVEL.
It works fine with VS2017.

sthalik, 2019-09-03:
Any objections to hardcoding it again?

Slion, 2019-09-06:
Hardcoding it again would just break both VS2017 and VS2019 builds.

Ideally we should report the issue to Microsoft once we can reproduce it in isolation.

Until this is fixed we are stuck with VS2017.

sthalik, 2019-09-22:
Why'd you be stuck? Is there some serious issue with MSVC 2019?

closed 2020-01-27 by sthalik

Visungky, 2021-11-02:
Hi, how do you solve this issue? I also met the same problem when I try to build from the source with MSVC 2019.

sthalik, 2021-11-03:
Build in release mode. In fact, never use the debug CRT.