opentrack/opentrack #981: wiimote + game data output crawl out of zero

Nasty76, 2019-08-23:
I get stupid problem which start it self no changing anythink
if imake head still and raw data show my head is position +-1 unit , output game data crawl or jumping over 15 units.

Nasty76, 2019-08-24:
i find reason for this , cos open track not have impelemnt camera treshold , some times open track not see dots , and when u open freetrack , treshold come on and then open track see dots clearly.
So if that hard job to make treshold slider in wiicamera too

cpuwolf, 2019-09-11:
impossible wiicamera is not working as you thought, wiiimote directly reports dots position

Nasty76, 2019-09-12:
well i know wii mote report only dots position BUT , my open track not see any dots , ok see if i go realy close my wiimote , and when i open freetrack and adjust treshold wiimote camera dadaaa open track see dots clearly :| so allways when i start play i need open opentrack then open freetrack and close freetrack.

Nasty76, 2019-09-12:
here is video https://youtu.be/L4rm8u-82Fk

sthalik, 2019-09-20:
Are you able to use the Wii sensor as a webcam in the freetrack example?

Nasty76, 2019-09-20:
No , freetrack has build in support for wiimote like opentrack but there IS Slider where you can adjudt treshold.and that report battery level too

sthalik, 2019-09-20:
@cpuwolf can you comment on that please?

closed 2020-01-27 by sthalik