opentrack/opentrack #983: FreeTrack not working with ED nor MWO

Slion, 2019-08-30:
I'm on the tip of master and can't get FreeTrack to work with ED nor MWO.
This was never a problem before. Did we change something?

Slion, 2019-08-31:
Rolling back to 1579a97c480578dcd14691cd3e40f56fd508204e fixes the issue.
TrackIR support was broken somewhere between those two commits.

Slion, 2019-08-31:
e9b5cd0e9dff9897b7f6c3e12c9563e3a40bf915 is working too.

sthalik, 2019-09-04:
Is this specific to that particular output method?

sthalik, 2019-09-04:
Can you see if 38d466a972d4455b02aee2b55674f3b29ece24d7 works please?

Slion, 2019-09-04:
66c2373014fc107d918b8549d554420d63405cc7 is working too and is more recent than the one you mentioned.

Slion, 2019-09-04:
Can't reproduce the issue anymore. Tip of master is working fine now, weird stuff.

Slion, 2019-09-05:
My bad, there is definitely something broken since 66c2373014fc107d918b8549d554420d63405cc7 .

Slion, 2019-09-05:
8fbcdf775660a0cc1857e8bedee7ac2351292aa6 is working too.

Slion, 2019-09-05:
7d898e638abad8267420f8c0d827d21f716b8977 is working.
03236ba81d6c85a26695968e5e4e266a18991362 is broken.

Slion, 2019-09-05:
9a308f07242d6ad996afdcd07c79abfe30293c8b is working.
Therefore 03236ba81d6c85a26695968e5e4e266a18991362 broke TrackIR support somehow.

sthalik, 2019-09-06:
Thanks for bisecting this. Does it work for you now? I'm not on Windows unfortunately.

Slion, 2019-09-06:
It's working now, thanks.

sthalik, 2019-09-22:
I wrote a test program, it's in the contrib directory.

closed 2019-10-12 by sthalik