opentrack/opentrack #985: Azure CI pipeline creation error

Slion, 2019-09-06:
I was getting the following error while trying to setup an Azure CI pipeline:

> Unable to configure a service on the selected GitHub repository. This is likely caused by not having the necessary permission to manage hooks for the selected repository.

Any chance you could give me the necessary permissions?

sthalik, 2019-09-07:
I gave you admin privileges.

closed 2019-09-07 by sthalik

Slion, 2019-09-07:
Thanks, that worked.
Though I can't find a way to install that GitHub application.
Is this working for you?
https://github.com/organizations/opentrack/settings/installations

sthalik, 2019-09-07:
It seems to be installed right now. What problem are you having?

sthalik, 2019-09-08:
Can you connect CI for this repository?

Slion, 2019-09-11:
Ok, fair enough, it looked like the request was still pending.

sthalik, 2019-09-15:
Looks like CMake configuration isn't setup yet. How much problem is it for you to enable all supported plugins from opentrack-depends and the like?

Slion, 2019-09-26:
Nothing is setup really. It's just an empty CI job at this point. I was hoping to setup a shared account but got stuck trying to resolve dependencies without providing a credit card number :)