opentrack/opentrack #986: Missing images in wiki

aristizabal95, 2019-09-09:
Some of the guides inside the wiki pages are missing images. This images are used to explain some of the procedures required to run the application correctly, and therefore are relevant for this pages. Here are the pages with missing content.

- https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)
- https://github.com/opentrack/opentrack/wiki/Smartphone-Headtracking
- https://github.com/opentrack/opentrack/wiki/VR-HMD-goggles-setup-----google-cardboard,-colorcross,-opendive

Len62, 2019-09-09:
The first link belongs to my work. It appears that dropbox is the culprit at this time. The videos show as they are on youtube. Have to research to see what went wrong with dropbox, the files are still there. Thanks for the heads up!

sthalik, 2019-09-11:
@Len62 can you upload the images directly into the wiki repository? Thanks!

Len62, 2019-09-11:
@sthalik Be glad to, just not sure how to do that. Is the repository a separate folder?

sthalik, 2019-09-13:
@Len62 you can clone it like a git repo (https://github.com/opentrack/opentrack.wiki.git) and push arbitrary files to it.

An easier solution is to ctrl+v a file or an image directly into the text input box.

aristizabal95, 2019-09-28:
Hello! This issue has been silent for a while. @Len62 if you wish I can try and upload the images to the wiki. I'd just need to have access to them.

sthalik, 2019-09-28:
@aristizabal95 the pages were created by different people. I'll try contacting some of them.

sthalik, 2020-01-27:
Unfortunately this is as much as I can do at this point.

closed 2020-01-27 by sthalik