opentrack/opentrack #988: How to build on Linux?

felixactually, 2019-09-24:
Sorry, I don't understand how to build this on Linux. Also, I want to use this for TrackIR in Euro Truck Simulator 2/American Truck Simulator. Will that work?

Any help appreciated.

felixactually, 2019-09-24:
I got it built but I don't know how to get it working with TrackIR.

felixactually, 2019-09-24:
I only have "FreePIE UDP receiver", "Fusion", "S2bot receiver" and "UDP over network" in input. None of the other input options in the getting started video are present.

felixactually, 2019-09-24:
I just tried a Snap I found via Discover, and that has PointTracker, and it doesn't say "no video", but the numbers don't change from 0 and it doesn't show anything.

sthalik, 2019-09-26:
> I got it built but I don't know how to get it working with TrackIR.

See the CMake cache variables with `cmake-gui` or `ccmake`.

felixactually, 2019-09-26:
I uninstalled Opentrack at this point, sorry. But what do you mean? Maybe I'll try again, I don't know.

sthalik, 2019-09-26:
There's a bunch of optional libraries you can specify in order to add functionality. That includes a LED-based tracker, VR device input, output for Wine games or a virtual joystick, etc. opentrack is modular enough that without additional libraries there's barely any functionality.

Termuellinator, 2019-12-13:
i can't build opentrack with aruco lib path selected in cmake-gui (complains about the include aruco/abi.h )
can you maybe point me in the right direction?

sthalik, 2019-12-15:
Have to use the https://github.com/opentrack/aruco fork.

Termuellinator, 2019-12-15:
yeah thats in the message, too.
Do i have to build that first or provide the source location? Building it fails (the original aruco builds fine) - but that is probably an issue for the aruco repo? ^^

When i build it without aruco, it builds fine, but has no input or output to select at all - that isn't normal, either, is it?
I would assume that i have to use some -DCMAKE flags that are not shown in ccmake/cmake-gui?

sthalik, 2019-12-15:
> Do i have to build that first or provide the source location? Building it fails (the original aruco builds fine) - but that is probably an issue for the aruco repo? ^^

I'd need the error message.

> When i build it without aruco, it builds fine, but has no input or output to select at all - that isn't normal, either, is it?

There's no circumstance where the software builds with no modules at all, after `make install`.

Termuellinator, 2019-12-15:
when trying to build opentrack/aruco with
mkdir build
cd build
cmake ..
cmake --build .

i get:
[a lot of lines with the undefined reference to]
/usr/bin/ld: /usr/lib/libopencv_viz.so.4.1.2: undefined reference to `vtkSmartPointerBase::vtkSmartPointerBase(vtkObjectBase*, vtkSmartPointerBase::NoReference const&)'
collect2: Fehler: ld gab 1 als Ende-Status zurück
make[2]: *** [utils/CMakeFiles/aruco_test.dir/build.make:151: utils/aruco_test] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:144: utils/CMakeFiles/aruco_test.dir/all] Fehler 2
make: *** [Makefile:130: all] Fehler 2

is opencv 4.1.2 the issue?

sthalik, 2019-12-15:
It shouldn't be an issue. Try building opentrack as-is. You already have `libaruco.a` at this point.

Termuellinator, 2019-12-15:
THANK YOU!
for the reference (for others maybe ;) )
cmake ..
ccmake . (selecting wine "on" and /aruco/build/src/libaruco.a from the "failed" build)
cmake --build .
make install

it now seems to work for me (at least the configuration and tracking inside the opentrack window)
Now on to see how well i get this to work with star citizen in wine ;)
Thank you again!

Termuellinator, 2019-12-15:
Well, it seems to work fine by itself, but starcitizen is not picking it up :/
Using "libevdev joystick receiver" works (kinda, probably need to finetune the curves a lot), but using "wine - windows layer for unix" does not.
I'm starting opentrack with "env WINEPREFIX="/home/termy/.local/share/lutris/lib/star-citizen/" ./opentrack " and set the wine in the output-settings to "/home/termy/.local/share/lutris/runners/wine/lutris-4.21-x86_64/" (the runner i am using in lutris), then start SC from lutris.
i don't have the "Freetrack 2.0 enhanced" output module that i see in many screenshots - is that normal in linux?
Star Citizen should work fine with opentrack in windows, so i think it must be an issue with the wine integration? I don't really found any way to troubleshoot what output is sent to wine - do you know of any demo or tool to test if it is working?

sthalik, 2019-12-16:
> i don't have the "Freetrack 2.0 enhanced" output module that i see in many screenshots - is that normal in linux?

Yes. Use the Wine module.

Termuellinator, 2019-12-16:
It still isn't working, but i think i've pieced some things together and/or have some questions:

- it is no longer necessary to start opentrack with "env WINEPREFIX="..." opentrack", but the prefix is set in the  wine output module settings after "wine (system)" , right?

-there is no way to tell opentrack to use another wine binary other than the systemwide wine, is there?
if not, that would be a nice enhancement request ;)

-is opentrack exposing trackIR and freetrack protocoll in wine? i saw a screenshot from windows where you could choose to only expose one or the other, so i hope trackIR is usable in wine

-is there any good way to troubleshoot why it is not working? I see the Freetrack entry in the prefixes user.reg registry, i'm using the same wine binary, but SC still doesn't react to the tracking. As it is working with opentrack on windows, i would assume that either my setup is wrong in some way or the wine protocol is doing something different than the windows protocol?

sthalik, 2019-12-16:
> it is no longer necessary to start opentrack with "env WINEPREFIX="..." opentrack", but the prefix is set in the wine output module settings after "wine (system)" , right?

Correct.

> -there is no way to tell opentrack to use another wine binary other than the systemwide wine, is there? if not, that would be a nice enhancement request ;)

Set `PATH`.

> -is opentrack exposing trackIR and freetrack protocoll in wine? i saw a screenshot from windows where you could choose to only expose one or the other, so i hope trackIR is usable in wine

Both.

> -is there any good way to troubleshoot why it is not working? I see the Freetrack entry in the prefixes user.reg registry, i'm using the same wine binary, but SC still doesn't react to the tracking. 

See `contrib/trackir-client`.

> or the wine protocol is doing something different than the windows protocol?

No.

Termuellinator, 2019-12-16:
> Set `PATH`.
i tried several combinations of launching it with the likes of:
`env WINE="/home/[...]/lutris-4.21-x86_64/bin/wine" WINESERVER="/home/[...]/lutris-4.21-x86_64/bin/wineserver" WINELOADER="/home/[...]/lutris-4.21-x86_64/bin/wine" ./opentrack` 
but it still seems to be using the system wine.
Am i missing something or did i misunderstand you?

BackOrder, 2020-01-12:
Which version of OpenTrack actually compiles on Linux? This would be a good starting point to fix the issues at hand.

From [OpenTrack 2.3.10 release](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3.10):
> Linux support isn't continuously tested, breaks often, and lags behind in features 

From **opentrack 2.4.0 release process** #778:
> #### Linux support
> summary of Linux support
> Built `unstable` rather than any particular stable release. It almost always has less bugs overall and is more likely to build at all.
> * [x]  beer Keybindings are expected to work fully on Linux but the key is "swallowed" by the window system while bound.
> * [ ]  beers There's halfway-done support for `.deb` packages but **don't expect it in the near future**. Once done I'll be able to generate packages for multiple Ubuntu versions.
> * [ ]  cry Linux support has lower priority (in features, bugs are fixed quickly in `unstable`) than Windows.
> * [ ]  cry Joystick input is not done. I volunteer to mentor you if you intend to work on this. Note, joystick **output** is fully functional and requires an in-tree kernel driver (included with some/most distributions).

[OpenTrack 2.3-rc99p14-snap1-gca299e4](https://snapcraft.io/opentrack) from Snappy package manager works but there are missing output, such as vJoystick. We might have some luck with contacting the maintainer, *Chen-Han Hsiao*.

Not compiling on **ArchLinux according to linux compilation error ( archlinux , gcc 9.2 , qt 5.13.1 , opencv 4.1 )** #726.

And I attempted to compile OpenTrack using 3 different compilers (gcc 9.2.0, gcc 7.3.1 from [Developer Toolset 7](https://www.softwarecollections.org/en/scls/rhscl/devtoolset-7/), clang 9.0.1) without success. Also using different flags such as C++ Standard Support 17, 14 and 11 without success.

Termuellinator, 2020-01-12:
i've had no luck with releases (only tried shortly though), but -git works well for me (if you know about using the libaruco.a after compiling that one failed ;) )

BackOrder, 2020-01-12:
> i've had no luck with releases (only tried briefly though), but -git works well for me (if you know about using the libaruco.a after compiling that one failed ;) )

Any instructions to follow? I had also used the current git repo to compile but without success. It seems that we have enough Linux users to collectively come up with something that will make the maintainers' life easier when it comes to releasing the Linux version.

Termuellinator, 2020-01-12:
See my previous comments ^^
```
git clone https://github.com/opentrack/aruco.git
cd aruco
mkdir build
cd build
cmake ..
cmake --build .
```
that fails for me, but the libaruco.a file is created.
then

```
git clone https://github.com/opentrack/opentrack.git
cd opentrack
mkdir build
cd build
cmake ..
ccmake . (selecting wine "on" and /aruco/build/src/libaruco.a from the "failed" build)
cmake --build .
make install
```
I can't **really** verfiy if it works - i only tried it with star citizen and that doesn't work unfortunately. Launching opentrack and using aruco to track seems to work well though, so it seems its either a wine-bug or something wrong with the wine-integration in conjunction with SC.

BackOrder, 2020-01-12:
> See my previous comments ^^

Sorry, mate! I did do my homework but it slipped through my fingers.

> ```
> git clone https://github.com/opentrack/aruco.git
> cd aruco
> mkdir build
> cd build
> cmake ..
> cmake --build .
> ```
> 
> that fails for me, but the libaruco.a file is created.

Aruco is actually building successfully here. I cloned OpenCV from github and then configured Aruco to use the newly built OpenCV. 

```
git clone https://github.com/opencv/opencv.git
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_C_COMPILER=clang -DCMAKE_CXX_COMPILER=clang++ ..
cmake --build . -- -j 8
```

Then follow the instructions you gave me but change `OpenCV_DIR` with `ccmake .` to use the build directory of the newly built OpenCV before building.

That's it for now. Still much to do.

BackOrder, 2020-01-12:
Compiling with either gcc or clang leads to just about the same kind of errors around `opentrack/options`. Anyone has any idea?

[OpenTrack-gcc.log](https://github.com/opentrack/opentrack/files/4051488/OpenTrack-gcc.log)
[OpenTrack-clang.log](https://github.com/opentrack/opentrack/files/4051489/OpenTrack-clang.log)

BackOrder, 2020-01-12:
Additional notes in regard to `opentrack/options`:

* [stable/2.3](https://github.com/opentrack/opentrack/tree/stable/2.3/options) compiles when using make in the `opentrack/options` directory.
* The changes affecting the compilation on Linux as far as `opentrack/options` goes are in [commit/f0216a3c53d43918295f1bd81975b391f4e5ed3](https://github.com/opentrack/opentrack/commit/f0216a3c53d43918295f1bd81975b391f4e5ed3b#diff-32ad475ed5ffd8a212c6e3a904828ea3) where maps are now initialized with {}.

sthalik, 2020-01-13:
> Compiling with either gcc or clang lead to just about the same kind of errors around opentrack/options. Anyone has any idea?

This should work in `unstable` with the possible exception of Qt 5.12.

sthalik, 2020-01-13:
I took the path of least resistance and require Qt 5.14 now.

BackOrder, 2020-01-13:
> I took the path of least resistance and require Qt 5.14 now.

Does it make any difference? My system uses Qt 5.9.7 and will be a challenge to upgrade.

BackOrder, 2020-01-14:
> I took the path of least resistance and require Qt 5.14 now.

I am not warming up to the idea after few hours building Qt and 20 GB disk space. Least resistance for whom again? :'(

sthalik, 2020-01-16:
> Least resistance for whom again? :'(

My sanity when reading template instantiation errors.

See if 9f41f6ee works.

BackOrder, 2020-01-17:
Hi @sthalik !

The latest changes up to 282fcb90dce76a55dc5f31246355fce2761a9eff you brought seem to be working now. How could I test OpenTrack without installing? Running `./opentrack/opentrack` will complain that some shared object files are missing. I put everything in a directory and opentrack opens but it won't find the input/output (which makes some sense).

Also, just FYI, I had managed to build Qt5 5.14.1, build OpenTrack and prepared a guide but it might be a lot of work for most people. Building alone takes few hours on an 8 cores i7 with SSDs.

sthalik, 2020-01-17:
Run `make install`.

BackOrder, 2020-01-17:
As a user? What's the default installation directory?

BackOrder, 2020-01-17:
Oh, yeah! Now we are talking!

Input and output are now populated and there is even libevdev joystick receiver.

[image]

sthalik, 2020-01-17:
How's that black color scheme? Is the mapping window background looking okay? The labels are probably invisible though.

BackOrder, 2020-01-17:
My desktop theme is dark. Let me know if you want me to test anything else. Might as well while we are at it!

[image]

[image]

[image]

[image]

sthalik, 2020-01-17:
@BackOrder great! It's always better for software quality to be familiar with uncommon configurations.

The mapping button image sticks like a sore thumb. The Octopus bottom shadow can't be helped since Qt doesn't support transparency in that case.

BackOrder, 2020-01-17:
For now, it seems that calibration is the most challenging. I haven't been able to successfully calibrate PointTracker. My webcam may or may not have IR (laptop, you know?), the LED lights lit up though. There are some things happening when I move my head but it doesn't correspond.

I did a side test with Aruco and printed a sheet of paper. It works. Can't imagine printing those and sticking it somewhere though. haha

FreePIE UPD Receiver only opens a port on UDP6 (IPv6) but not on UDP (IPv4) ??

> The mapping button image sticks like a sore thumb. The Octopus bottom shadow can't be helped since Qt doesn't support transparency in that case.

I suppose it's not a big deal and it might be due to my theme.

BackOrder, 2020-01-17:
For reference with a white theme:

[image]

sthalik, 2020-01-27:
> For now, it seems that calibration is the most challenging. I haven't been able to successfully calibrate PointTracker. My webcam may or may not have IR (laptop, you know?), the LED lights lit up though. There are some things happening when I move my head but it doesn't correspond.

Did you set up the model? Unfortunately I don't have the resources to provide basic user support. There are *many* tutorials online, on YouTube and otherwise.

closed 2020-01-27 by sthalik