opentrack/opentrack #989: DCS

FernandoNotedigo, 2019-09-24:
Dear all,
I've installed this with PS3 camera for DCS.

At first it worked but after a reset on the PC it seems the program is not sending the data to DCS or DCS is not getting it. When I press "start" i can see the opentrack getting the 3 leds of my clip and showing game data numbers but i don't know if DCS is getting them. 

I don't know how to debug it. Could you help me?

Thanks.

Len62, 2019-09-24:
What settings are you using for input, output, and filter?

FernandoNotedigo, 2019-09-25:
> 
> 
> What settings are you using for input, output, and filter?

Input: PointTracker 1.1 (only thing touched was the calibration)
Output: freetrack 2.0 Enhanced (configured for FreeTrack, hide TrackIR)
Filter: Accela (untouched)

I've continued touching things and noted that sometimes I get it working for one time going to "Locate DLL" and selecting NPClient.dll from the directory. I don't know if this is chance but is the only thing I was able to modify that has some effect.

Thank you!

Len62, 2019-09-25:
One thing to try is to enable both  FreeTrack and TrackIR in the output settings.

FernandoNotedigo, 2019-09-25:
> One thing to try is to enable both FreeTrack and TrackIR in the output settings.

I've tried that with no effect. I wonder if there is some log in some place I could use to debug.

sthalik, 2019-09-26:
> I've continued touching things and noted that sometimes I get it working for one time going to "Locate DLL" and selecting NPClient.dll from the directory. I don't know if this is chance but is the only thing I was able to modify that has some effect.

This shouldn't make a difference. Does your Windows user name contain diacritic characters?

sthalik, 2019-09-26:
> I've tried that with no effect. I wonder if there is some log in some place I could use to debug.

See if this registry key exists and has the right value: `Software\\NaturalPoint\\NATURALPOINT\\NPClient Location`.

FernandoNotedigo, 2019-09-30:
> 
> 
> > I've tried that with no effect. I wonder if there is some log in some place I could use to debug.
> 
> See if this registry key exists and has the right value: `Software\\NaturalPoint\\NATURALPOINT\\NPClient Location`.

Hi, sorry for the delay. The key exists but values are empty. 
I have (Default) and Path values to fill. I guess I should place the path into the Path field. Right?

Thank you!

sthalik, 2019-10-12:
Select "enable both" in freetrack protocol settings. Right now you have "hide trackir" selected and that's causing the problem.

closed 2020-01-27 by sthalik