opentrack/opentrack #991: Shortcuts

Verc0rs, 2019-09-29:
Sometime the shortcut selected on my keyboard or joystick don't work, it appear in the window shortcut but i have to enter it again to make opentrack recognize it. 
thanks

MathijsG, 2019-10-02:
Did you plug it out of the USB port and in another USB port in between those cases?

Verc0rs, 2019-10-02:
I "fix" the problem. It was because I use the same shortcut for opentrack and my game (dcs world):
- shift + button 1 for center the view in opentrack
- shift + button 1 for view in the cockpit in DCS
Most part of time it's work fine and sometime that create a conflict.

sthalik, 2019-10-12:
How many CPU cores do you have? How many cores does the game use?

Verc0rs, 2019-10-12:
intel k3770, 4 real cores and 4 simulated cores. But I don't know the number of cores used by my game.

sthalik, 2019-10-21:
Does this happen if you hold the key longer, i.e. for half a second rather than only tapping it?

Verc0rs, 2019-10-22:
No it doesn't appear if I hold the Key longer.
Doing some few tests, I notice the problem alway appear when I launch and select an application named "DCS-SRS client" (used for the communications in the game "DCS world").
If "DCS-SRS client" is selected, the shortcut mapped in "Opentack" don't work. The problem disappear when I select "Opentrack" again or another application ( try with "VLC" and "Firefox"). So it's maybe a problem of "DCS-SRS client" and not "Opentrack".
For the moment, the problem doesn't reappear in game like I notice the first time (even with two commands for the same shortcut).

sthalik, 2019-10-22:
> No it doesn't appear if I hold the Key longer.

Open Task Manager and assign realtime priority to the opentrack process. See if this still happens.

Verc0rs, 2019-10-23:
If you mean "try with hold the key longer", then there is no difference.

If you mean with "DCS-SRS client" selected, it still happens with "Opentrack" in realtime priority, and even if I assign "DCS-SRS client" in low priority.

sthalik, 2019-10-30:
I can't reproduce it, and looking at the code I'm not sure how that could possibly happen. Note that you need the exact set of modifier keys that you used when creating the keybinding. That is, binding F11 won't trigger it with shift+F11. But that probably doesn't help you at all.

sthalik, 2019-11-01:
> I use "antimicro" to assign keyboard keys on my sidewinder ffb2 joystick.

That might be the problem.

Verc0rs, 2019-11-04:
OK, I will try to look a this.

Verc0rs, 2019-11-25:
It's between antimicro or windows, I'm not sure, but not opentrack.

closed 2019-11-25 by Verc0rs