opentrack/opentrack #1002: Octopus not moving

sutex, 2019-11-21:
Set it up last night played Star Citizen and today, opentrack is not working in game, the Octopus is not moving either, the LED are showing

sthalik, 2020-01-27:
Not an opentrack issue.

sutex, 2020-02-13:
Opentrack works when setting up a new profile (ini) to use with Star Citizen, until I close Opentrack. 
When relaunching Opentrack that same profile refuses to work, the octopus wont move and I have to redo a new Profile. This is happening regularly.

sthalik, 2020-02-13:
There's nothing in opentrack internals that could be causing that issue. Creating profiles doesn't affect games in any way.

sthalik, 2020-02-13:
I'll reopen this issue for several week in case a community member can provide support.

sutex, 2020-02-13:
It's not the game.  Happens with any profile in the list. As soon as I change setting in a Profile, use it then close down. That profile refuses to work when starting back up , along side any other Profile ini in the lis. Only way I can get a Profile to work is to create a new one

sthalik, 2020-02-14:
I didn't see the part about the octopus not moving. Send me your profile (ctrl+v the file itself into this text box, not the content).

sutex, 2020-02-14:
I included all the Profiles. the only one working is SC.ini all others the octopus is not moving
[Profiles opentrack-2.3.zip](https://github.com/opentrack/opentrack/files/4202204/Profiles.opentrack-2.3.zip)

sthalik, 2020-02-14:
Which is the first thing that doesn't work? Do you get raw data? If so, do you get a red dot on the mapping showing the input position? Does PT tracker detect the points at all?

It might be helpful to launch opentrack inside a Windows console (drag-and-drop `opentrack.exe` onto an open console window). In case of several internal failures you're going to get additional information.

sutex, 2020-02-16:
Not sure about that list, I'll explain what I see 
The camera LED pixels points move.
There is no raw data / game data showing, all the values are zero
Mapping curves nothing moves
Octopus does not move.

sthalik, 2020-02-18:
Paste a screenshot to the main interface then.

sutex, 2020-02-18:
[image]

[image]
[image]

[image]

sthalik, 2020-02-21:
The raw data looks like it failed to compute the pose despite having the data available. Open Windows console, drag the executable onto the console window, then execute in order to get a basic debug log. You should get an error message like "POSIT failed".

sutex, 2020-02-21:
The exe in the console window and hitting enter which opens Opentrack
C:\Program Files (x86)\opentrack>
DEBUG [C:\Users\Administrator\dev\opentrack\api/plugin-support.hpp:107]: library "kinect-face" failed: "Cannot load library C:\\Program Files (x86)\\opentrack\\modules\\\\opentrack-tracker-kinect-face.dll: The specified module could not be found."
WARN DirectWrite: CreateFontFaceFromHDC() failed (Indicates an error in an input file such as a font file.) for QFontDef(Family="MS Shell Dlg 2", pointsize=8.25, pixelsize=11, styleHint=5, weight=50, stretch=100, hintingPreference=0) LOGFONT("DejaVu Sans", lfWidth=0, lfHeight=-11) dpi=96
 
........................
Starting the Track with Profile SC.ini output
[ INFO:0] global ..\modules\videoio\src\videoio_registry.cpp (187) cv::`anonymous-namespace'::VideoBackendRegistry::VideoBackendRegistry VIDEOIO: Enabled backends(3, sorted by priority): DSHOW(1000); CV_IMAGES(990); CV_MJPEG(980)

sthalik, 2020-02-22:
Ah, I see it. In `SC.ini` you have `active-model-panel` set to 0. In the other files you have it set to 2, aka the custom model (which has all points set to one value). Select the first tab (i.e. the cap) and confirm whether it fixes the issue.

sutex, 2020-02-24:
I'm running a DELANCLiP and SC.ini is the one working.

sthalik, 2020-02-25:
You got your answer. The PT tracker is misconfigured:

> Select the first tab (i.e. the cap)

sthalik, 2020-02-25:
Calibration only makes sense once tracking works.

sutex, 2020-02-25:
That not making sense SC.ini is working so why would choose 'Cap' instead of the Clip tab when I'm running the Delan Clip?

sthalik, 2020-02-25:
It's fine in SC.ini but not elsewhere.

sutex, 2020-02-25:
Ok I think your not getting what's happening, I set a profile, as it is in SC.ini , run in game for awhile, then next day or few simes used it refuses to work. As all the other Profile in that Zip file are showing, they at one stage did and where setup as SC.ini is now.

sthalik, 2020-02-26:
As I keep telling you, all your other profiles have the Custom tab selected and that won't work.

sthalik, 2020-02-26:
That is incorrect:

```
[tracker-pt]
active-model-panel=2
```

That is correct:

```
[tracker-pt]
active-model-panel=0
```

sutex, 2020-02-26:
I know you keep telling me about the other profiles, forget about them. They where setup as is SC.ini is now, yet the [tracker-pt] changed. So why is Opentrack changing the [tracker-pt] profiles when all I'm doing is shutting down, then restarting the next time I'm about to play.

sthalik, 2020-02-26:
> So why is Opentrack changing the [tracker-pt] profiles when all I'm doing is shutting down, then restarting the next time I'm about to play.

The settings can't change unless the dialog itself is opened. Otherwise there's nothing that saves the dialog contents. I can point you to relevant code. If that's incorrect, I still need steps to reproduce it.

sutex, 2020-02-28:
All I can do is tell you what is happening and give the the profiles and information I have.

sthalik, 2020-02-29:
OK, show me the model configuration tab.

sutex, 2020-02-29:
[image]