opentrack/opentrack #1008: UDP over network output frequency

JimmyRaoUF, 2019-12-10:
Hi, I am having a delay problem using UDP to send filtered data into Unity (I wrote a receiver in it without limiting the frame rate (update frequency)), not sure why maybe because of buffer?
I am thinking because the Kinect/Camera runs at 30Hz, I can just limit everything to 30Hz.
So is there a set frequency/default frequency for UDP output in OpenTrack?

Thanks~

sthalik, 2019-12-13:
> So is there a set frequency/default frequency for UDP output in OpenTrack?

It's 250 frames a second.

JimmyRaoUF, 2019-12-14:
@sthalik Thank you sthalik! What will happen to the between frame data if the tracker is 30 Hz? Will it just repeat itself?

sthalik, 2019-12-14:
Filter will be applied each time.

closed 2020-01-08 by JimmyRaoUF