opentrack/opentrack #1010: Oculus startup failure

MikeRomeo8801, 2019-12-16:
When I set the input to "Oculus Rift runtime 1.4.0 -- HMD" I get the message: 

"Error occurred while loading tracker Oculus Rift runtime 1.4.0 -- HMD

ovr_Create: No HMD attached."

I reinstalled the oculussoftware yesterday to use my Oculus rift again after a half year and it worked fine. I also wanted to try out whether I could use the Oculus sensors and opentrack to get a reasonable head tracker up and running but it didn't seem to work. I only get this error message and the Oculus software opens in the background.

Did I do something wrong or is it due to open track? Do I have to connect both sensors or only one?

christoschronopoulos, 2020-01-27:
I get the same error. In Oculus Home app the sensor (dk2) seems to be connected but when I press "Start" in opentrack the following appears 

> **_Error occurred while loading tracker Oculus Rift runtime 1.4.0 -- HMD
ovr_Create: No HMD attached._**

JawDamn, 2024-11-05:
same here tracker Oculus Rift runtime 1.4.0 -- HMD Unable to connect to service: NotFound: Server endpoint not found