opentrack/opentrack #1011: Valve/SteamVR: HTC Trackers Not Detected; Lighthouses Marked as "Tracker"

Shanesan, 2019-12-16:
Using OpenTrack-2.3.11 beta3, but results are the same as on the GitHub.

Issue: HTC Tracker (the "puck") is not detected by software.

[Image 1](https://i.imgur.com/Ot5ZuML.png)

In Image 1 you will see equipment marked as "HMD", "Tracker" and "Tracker". The HMD is correct. The "Tracker" is an HTC V2-XD/XE which are the lighthouses. The Tracker that is on (LHR-19DCC091) is not detected.

Shanesan, 2019-12-17:
I saw there was discussion on this in Issue #352, specifically @sthalik's comment [here](https://github.com/opentrack/opentrack/issues/352#issuecomment-322386981) asking for a screenshot, provided in Image 1. Please let me know if you need anything else from me on this issue!

Wesloaf, 2020-01-20:
I am having the exact same issue, I imagine it has to do with the way steam vr changed how the trackers are labeled, like how you have to specify if its held in hand or is tracking a certain joint. I would like to see this fixed though.

sthalik, 2020-01-27:
> Just a note how "Pucks" work, as we know only two wireless devices can hook into the headset, typically the left controller and right controller. The pucks hook up to a USB dongle. You can force it to pair with the headset itself but this does not alleviate the issue.

Are these the controllers the person is holding in this video? <<https://www.youtube.com/watch?v=qplccIU3ybE>>

The openvr glue code requests controllers from a sitting position. Using the standing position results in wrong centering.

Shanesan, 2020-01-28:
Yes those are the same devices (the one attached to the racket, et al). Sorry for my ignorance, how would I configure it from a “sitting position” to test if this detect the controller successfully?

relu-ai, 2021-05-04:
Hi, got the same issue here. Using a Lighthouse 1.0 and a vive tracker 2.0.
The base station is appearing as tracker and the tracker itself is not working. 

Happy to help test where I can, would be an excellent use case as a TrackIR alternative

Shanesan, 2021-05-04:
I have also done some additional testing on this, got a Tracker 2.0 (blue button) and THAT tracker is detected, but the Tracker 1.0 (gray button), so I think there's something wrong with the detection in OpenTrack for SOME trackers.

relu-ai, 2021-05-04:
@Shanesan are you able to track using the Tracker 2.0 with base stations as the only tracked item (e.g. no controllers or hmd)?

proficiency, 2022-07-21:
this issue persists today, opentrack 2022.3.0 is unable to detect my v1 trackers & lighthouses appear as trackers in the device list.
[image]

apologies for bumping a 3 year old issue but it's really annoying duct-taping a vive controller to my headphones when these nice little pucks are around

Shanesan, 2022-07-21:
To answer @relu-ai (I gave up a long time ago, thanks @proficiency for bumping) the HMD needs to be on on the HTC VIVE but no controllers are needed.

sthalik, 2022-07-21:
Run opentrack from Windows Console (drag the executable or the shortcut onto a console window, then press Return), then see whether there are any interesting messages when you open the tracker settings window. That's where enumerating the devices takes place.

Currently there's no logic in opentrack that would filter out devices. If it's being returned by OpenVR, it should be available to you in opentrack.

proficiency, 2022-07-21:
@sthalik 
seems like this is the only thing out of the ordinary
DEBUG [F:\dev\opentrack\tracker-steamvr\steamvr.cpp:86]: steamvr: getting model number failed for 4