opentrack/opentrack #1016: Error opening PS Eye camera on Linux

paju1986, 2019-12-29:
Error log when traying to start the tracking on ps eye camera on pointracker.

WARN Icon theme "gnome" not found.
DEBUG [/home/alberto/proyectos/opentrack/qxt-mini/qxtglobalshortcut.cpp:58]: qxt-mini: adding event filter
DEBUG [/home/alberto/proyectos/opentrack/qxt-mini/qxtglobalshortcut_x11.cpp:359]: qxt-mini: fixed x11 autorepeat
Icon theme "gnome" not found.
DEBUG [/home/alberto/proyectos/opentrack/logic/runtime-libraries.cpp:25]: protocol dylib load failure
WARN QXcbConnection: XCB error: 3 (BadWindow), sequence: 2403, resource id: 35651807, major code: 40 (TranslateCoords), minor code: 0

paju1986, 2019-12-29:
Close, i didnt choose a output library that was the error

closed 2019-12-29 by paju1986

morse-bip-bip, 2021-07-17:
Hi ! i'm having the same issue but I don't understand how you fixed it. How do you choose an output library ? thanks