opentrack/opentrack #1017: cmake should detect minimum required QT Version

rvt, 2019-12-29:
cmake should detect minimum required QT Version.
Since we removed our own hash template for QString we now get this odd error (if you know it's easy to fix..)

However we should detect in cmake that the minimum required version of Qt is now 5.14.

sthalik, 2019-12-29:
Works fine on CI with Qt 5.11. Can you reproduce on `unstable` branch?

rvt, 2019-12-30:
Most be something I overlooked.

When I change locally from qt 5.13 to 5.14 Build failed because the template already exists `template<> struct hash<QString> { ....`

therefor I though the template was added by Qt on release 5.14. Must have been a side effect of something else? CLosing out..

closed 2019-12-30 by rvt