opentrack/opentrack #1020: P3P solver in Kinect not accurate on Z-depth and also pitch and yaw

JimmyRaoUF, 2020-01-08:
Hi, thanks for the amazing work you provided here,
I almost succeeded in using Kinect IR marker solution (Easy Tracker version) here,
with some little buggy stuff I can't solve.
The P3P solver in Kinect not accurate on Z-depth and also pitch and yaw,
I am guessing it's related to the cap/IR marker model?
I tried these 2 set up, one bigger and one smaller:
[image]
[image]

Both of them have a similar performance issue in Easy Tracker.
Thanks in advance!

Slion, 2020-01-08:
I thought you got it working eventually #971 

Which OpenTrack version are you using now? Make sure you are not too close from your Kinect. I'll come back to you with my config. It used to work just fine.

Slion, 2020-01-09:
Here is what I'm using:
[image]

It's working well enough it seems though I believe the octopus is being funny at times so make sure you try it in your application too or check raw and game data rather than just looking at that octopus.

I find Z offset working really well actually when being around 70 cm from the sensor.
Admittedly the solver is sometimes being funny with extreme positions like looking at the bottom right corner for instance but TBH there should be no much need for such extreme head movements.
You may also want to try different positioning of your cap on your head and recenter to see which one is giving best results for your application.

Slion, 2020-01-13:
Also I might be considering using the following solver at some point:
https://github.com/midjji/lambdatwist-p3p

JimmyRaoUF, 2020-01-13:
Thank Slion, I look forward to the version of the new solver.

whiskeygreg, 2020-08-28:
It seems I am also having this problem. My side to side movement is interpreted as forward and backward.

Slion, 2020-08-28:
> It seems I am also having this problem. My side to side movement is interpreted as forward and backward.

Make sure your vertices coordinates are correct. Check the coordinates signs are similar to the above. If still having issues try putting the vertices further appart on your hat and adjust the coordinates accordingly.

whiskeygreg, 2020-08-28:
Slion,

Thanks for the response. The colors are indeed matched. I have the latest version of opentrack, 2.3.12, and am using Easy Tracker 1.0. I matched the numbers above exactly, and my hat is set up identically, though my points are bigger.

Slion, 2020-09-01:
There is a test version available, see:
https://github.com/opentrack/opentrack/issues/1141#issuecomment-684961261

PR #1143 (2020-09-02): tracker/easy: Disable Kinect distortion coefficients usage
Thus notably fixing badly broken Z component in translation vector.

Close #1141
Close #1020
commit e25f8189 (2020-09-02): tracker/easy: Disable Kinect distortion coefficients usage thus notably fixing badly broken Z component in translation vector.
Close #1141
Close #1020
closed 2020-09-02 by Slion