opentrack/opentrack #1023: Odd "stuttering" in programs like Virtual Desktop and VR Toolbox when locking screen to head

efhunter, 2020-01-12:
In VR desktop programs like Virtual Desktop and VR Toolbox: 360 Desktop, opentrack works fine with them until setting the virtual display in either program lock to the user's head (with no delay).

The issue is very odd: having opentrack track at the same time causes the **_physical movement of the virtual display_** to update its position in a stuttery manner. The **_image on the virtual display itself updates perfectly fine_**, in a smooth manner, as one would expect.

Being able to lock the virtual display to one's head (with no delay), coupled with opentrack would allow for one to look around as much as they want without ever looking away from the screen whatsoever. It was quite surreal doing this in ArmA 3 apart from the stuttering problem.

Need4SpeedOG, 2020-01-14:
I'm seeing a similar issue after updating 2.3.11 from beta 3 to release. My input is from a RealSense camera tracking was smooth in the beta now I get wild fluctuations with the slightest. I've flipped between the beta and release and there appears to be a input delay occurring during processing of the raw data. The delays are very noticable in the logging data I'm attaching here. 

[opentrack-2.3.11-beta3-logging-accelafilter.xlsx](https://github.com/opentrack/opentrack/files/4057040/opentrack-2.3.11-beta3-logging-accelafilter.xlsx)
[opentrack-2.3.11-release-logging-accelafilter.xlsx](https://github.com/opentrack/opentrack/files/4057041/opentrack-2.3.11-release-logging-accelafilter.xlsx)

sthalik, 2020-01-16:
> physical movement of the virtual display to update its position in a stuttery manner

What does this mean?

efhunter, 2020-01-16:
> > physical movement of the virtual display to update its position in a stuttery manner
> 
> What does this mean?

The floating virtual monitor in Virtual Desktop/VR Toolbox can be locked to your head. So, it physically moves with your head, whenever you turn your head. When having the floating virtual monitor be locked to your head with no delay, as soon as opentrack starts tracking, the movement of it becomes a lot choppier.

It's strange, because the image on the floating virtual monitor updates smoothly, despite this choppy movement.

sthalik, 2020-01-16:
> It's almost like opentrack's tracking is somehow interfering with headtracking the programs are doing, 
> and they start receiving headtracking input at a much slower rate, but I have absolutely no clue.

Both of the apps are using the same tracking input, correct?

sthalik, 2020-01-16:
opentrack gets data at 250 Hz. That may be starving other processes.

sthalik, 2020-01-27:
Unfortunately there's nothing I can do here. HMDs use high refresh rates so polling at reduced speed isn't an option.

closed 2020-01-27 by sthalik