opentrack/opentrack #1024: Buttons to axis

Ec0nomist, 2020-01-12:
There are unofficial versions opentrack, which allow you to control the displacement axes keyboard or joystick buttons. You plan to add this functionality officially?

sthalik, 2020-01-13:
That falls outside of what opentrack is made for -- I could be persuaded otherwise.

Ec0nomist, 2020-01-17:
This is true for some users of the inertial tracker. Now it looks like this:
https://i.ibb.co/tCgb7Cx/Readme.jpg
I can send you this version

sthalik, 2020-01-17:
It's not mergeable in its current form. It should be made with a tracker. Trackers can already make their own keybindings. Fusion allows for combining trackers for rotation and translation.

closed 2020-01-27 by sthalik

Gianpaolo79, 2020-03-26:
> There are unofficial versions opentrack, which allow you to control the displacement axes keyboard or joystick buttons. You plan to add this functionality officially?

Where can I find this version?

Ec0nomist, 2020-03-26:
> 
> 
> > There are unofficial versions opentrack, which allow you to control the displacement axes keyboard or joystick buttons. You plan to add this functionality officially?
> 
> Where can I find this version?

https://drive.google.com/open?id=14huWP1AK_snoS9JITg13R2Qh5092M6ao

Gianpaolo79, 2020-03-26:
Thank you! 
Do you know why it does not see my Ps3eye camere while the standard version does?

Ec0nomist, 2020-03-27:
> 
> 
> Thank you!
> Do you know why it does not see my Ps3eye camere while the standard version does?

I use inertial tracker, sorry