opentrack/opentrack #1030: vjoystick driver problem

KulaGGin, 2020-01-15:
When I press on the Start button, it shows this error:
https://i.imgur.com/ciOYHSz.png
And then this error right after it:
https://i.imgur.com/3UekqiW.png

This is after I pressed on Download the driver button, restarted PC and tried again. Reinstalled vjoy driver just few minutes ago:
https://i.imgur.com/VVARHlW.png

Works in vJoy feeder and FreeTrack No IR just fine.

KulaGGin, 2020-01-16:
@sthalik Ok I cloned this repo and I'm on unstable build. How do I compile it?

KulaGGin, 2020-01-16:
Looks like there are a lot of dependencies. Can you guy compile and post binaries of the last build on the unstable branch, so we can investigate further why vjoy doesn't want to work on my system?

I'll try to compile and try by myself but it can take me quite a while for the first time and I will just might get into other problems with compilation, so I'll have to create another issue for that...

sthalik, 2020-01-17:
This isn't a fix -- yet.

KulaGGin, 2020-01-18:
I understand. I thought it would help if it logged version numbers and you would see what version it shows in my case

ForceLightning, 2020-01-23:
I am also facing this issue with vjoy 2.1.9.1 and opentrack experimental build 20191023.
[image]

Demuth92, 2020-01-24:
Same issue, any solution yet ?

sthalik, 2020-01-27:
See if today's build from <<http://opentrack.misaki.pl>> works.

ForceLightning, 2020-01-28:
Today's build works. Thanks.

closed 2020-01-28 by sthalik