opentrack/opentrack #1031: Linux: VIDEOIO ERROR: V4L: index 0 is not correct!

fxkr, 2020-01-18:
I've built the latest opentrack (incl changes from #988 - many many thanks to @sthalik ) on Fedora 31.

When I click "Start", I get an error and the video does not start:

```
VIDEOIO ERROR: V4L: index 0 is not correct!
```
Looking at `video-opencv/impl-camera.cpp` `cam::start` line
```
cap.emplace(idx + video_capture_backend);
```
I added some debug logs: `idx` is 0 and `video_capture_backend` is 200.

Through `ls /dev/video*` and unplugging/replugging my camera, I have found that my USB webcam is `/dev/video1`.

So I have experimentally tried passing 201. And with that the video starts successfully and tracking works.

Any tips how to fix this correctly?

This is a Thinkpad T480s which has a builtin cameras (normal + IR), but I'm using a separate USB webcam.

```
% ls /dev/video*
/dev/video1  /dev/video2  /dev/video3  /dev/video4  /dev/video5  /dev/video6
```

sthalik, 2020-01-27:
Confirm if it works on your end, thanks!

fxkr, 2020-01-28:
Works, thanks!

closed 2020-01-28 by fxkr