opentrack/opentrack #1033: Mapping pitch axis related to yaw axis

rogamaral, 2020-01-19:
I don't know if it is easy to implement or if it is already possible to do it with mapping. What I want is when I look forward, through the windshield, any movement of the head down would be ignored. I have a home made cockpit and I don't want the virtual cockpit to appear on the screen when I look forward and down. But when I look out the side windows, I want to see up and down.

sthalik, 2020-01-27:
That's too specialized to implement.

closed 2020-01-27 by sthalik

rvt, 2020-01-30:
Could we allow people to create lua scripts for these specialisations? Then they can decide what to do under specific circumstances..

sthalik, 2020-01-31:
@rvt yes! There's already some commented out extension support (but no bindings). The basic idea is letting the user modify the result after each pipeline stage. I can't work on it in any reasonable timeframe though.