opentrack/opentrack #1034: PS3 Eye is bugged out in opentrack

KulaGGin, 2020-01-19:
I used CL-Eye-Driver-5.3.0.0341-Emuline driver. That's the only driver opentrack even sees the PS3 Eye camera with. It doesn't see PS3 eye with other drivers at all. I tried IPIsoft PSEYE Driver Installation and Libusb PSEYE Driver Installation from here: https://github.com/psmoveservice/PSMoveService/wiki/PSEye-Software-Setup-%28Windows%29

Ok, so back to the CL-Eye driver. When I open opentrack and press on Start, it shows cut double image at low FPS. Works fine in CL-EyeTest.exe. I recorded a video to show what happens:
https://youtu.be/4J6iz1u0MTY

Can someone recommend me  some driver that will work in opentrack?

I'm on this Windows version:
https://i.imgur.com/jFZG1rV.png

Len62, 2020-01-20:
In the cl-eye test preview why is the camera showing your screen? Using the pointtracker for input, the camera would need to be aimed at your pointracker, whatever type you are using.

KulaGGin, 2020-01-20:
Ok, I just installed Windows 7 on a drive, and PS3 Eye webcam works fine on Windows 7 in CL-Eye Test and in opentrack on a CL-Eye-Driver-5.3.0.0341-Emuline driver. So it's something-something driver issue. But still, PS3 Eye works in a test program but opentrack glitches out on Windows 10.

Len62, 2020-01-20:
My PS3 driver is version 5.3.0.0341 from Code Laboratories and I have no problems running OT in Windows 10. Which version of OT are you running?

KulaGGin, 2020-01-20:
Last version - [opentrack  2.3.11](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3.11).
Which version of Windows 10 are you running? I'm on 1709:
[image]

Len62, 2020-01-20:
Can you do a video like the one you posted but with everything set-up to run OT? I'm interested in seeing the output in cl eye driver test and then also what OT is showing. 
[image]

KulaGGin, 2020-01-21:
Well, I decided to reinstall the system.
I backed everything up, installed the exact same Windows version(I tried it first on a drive with free space) and it works:
[image]

I wouldn't say that my system was wrong, it wasn't bloated, not really: I had a lot of useful stuff installed like Visual Studio, Atmel Studio, Solidworks, Photoshop, Office, VMWare Workstation, Premiere Pro, UE4 Engine, and many other programs, but these were the biggest ones that installed the most dependencies like .NET frameworks, C++ redistributables, SQLs, etc.
I'll keep an eye on opentrack and will try to track what caused this issue. If I find what cuased it, I'll report here I think. My bet that it's some C++ redistributable or something of some version that some program installed and it caused conflict in opentrack for whatever reason.

Len62, 2020-01-21:
Great! Glad you got it sorted out.

closed 2020-01-22 by KulaGGin

sthalik, 2020-01-27:
> If I find what caused it, I'll report here I think.

Did you install any codec packs or "cleaners"?