opentrack/opentrack #1035: Linux: Wayland: Keyboard shortcuts don't work, alternatives?

fxkr, 2020-01-20:
Wayland doesn't allow applications to register global shortcuts anymore. On Xorg opentrack's shortcuts still work, but on Wayland shortcuts don't work and you get this error message:

```
WARN [.../opentrack/qxt-mini/qxtglobalshortcut.cpp:93]: QxtGlobalShortcut failed to register: QKeySequence("Print")
```

The generic way around this seems to be adding an "API" to the program (opentrack, in this case) that can be called from the command line, and asking users to configure a shortcut in their window manager (e.g. GNOME) that calls this API.

A github issue referenced by other programs with the same issue -> albertlauncher/albert#309

sthalik, 2020-01-27:
> The generic way around this seems to be adding an "API" to the program (opentrack, in this case) that can be called from the command line

No way.

> Wayland doesn't allow applications to register global shortcuts anymore.

There was never any support for native Wayland. I assume this was done through XWayland. Try using x11 Qt QPA.

closed 2020-01-27 by sthalik

fxkr, 2020-01-28:
> There was never any support for native Wayland. I assume this was done through XWayland. Try using x11 Qt QPA. I'd consider native support if there's any adoption. Probably not in several years.

fwiw Wayland is the default in Debian(-if-using-GNOME) and Fedora.

Anyway, now that I'm aware of what the issue is I'm fine; I can just run opentrack in Xorg. I created this issue mostly for the benefit of the next user. Maybe this should be in the common issues list in the wiki.

(I don't even disagree with your decision, "no way" is what I thought when I read about the global hotkeys situation in Wayland as well.)

sthalik, 2020-01-28:
Does it help any if you use `opentrack -platform xcb` with Xwayland?

fxkr, 2020-01-29:
With `opentrack -platform xcb`, the warning is gone, and the shortcut works while the opentrack window is focused, but not when another application is focused.

(fyi, with Xorg there also was another warning and that goes away too with `-platform xcb`:  `WARN Wayland does not support QWindow::requestActivate()`)

XenoPL, 2022-10-25:
Bump on this matter, wayland becomes default for both gnome and kde in most of the popular distros in 2022. I'd love to have  working shortcuts,quite handy thing for neuralnet tracker which by default is not centered along z-axis by default.
How difficult/work intensive  would it be to make it work?

sthalik, 2022-10-26:
I'm definitely not going to target each compositor separately. At this rate it might be easier to read the keyboard's raw USB events and demand that the user sets up permissions through udev.

JT8D-17, 2023-08-10:
`opentrack -platform xcb` fixes the non-working shortcuts withn Wayland for me in KDE Plasma 5 on Arch Linux. Even when opentrack is not in focus.

There also is "Legacy X11 app support" in the Plasma settings menu ("Applications"), which enables keyboard shortcuts passthrough, but this did not seem to have any effect on opentrack for me.

krims0n32, 2024-07-26:
Any updates on this? Makes Opentrack in Wayland cumbersome because keybinds do not work. -platform xcb works but X apps in Wayland are becoming a pain in the ass compatibility wise. Passthrough functions (KDE, Hyprland) do not work either so there isn't a workaround really when running Opentrack in native Wayland mode, is there?