opentrack/opentrack #1038: Constant Flipping issue

JimmyRaoUF, 2020-01-29:
Hi, I am using Kinect with OpenTrack and Easy Tracker,
recently (this never happened before), when I try to track,
it constantly flips, so never stable tracking.
I have attached the video on my case here:
[https://youtu.be/fP1CnsXztF0](url)

Thanks!

JimmyRaoUF, 2020-01-29:
Is it related to the auto center?

JimmyRaoUF, 2020-01-29:
Seems might be related to auto center. (in the input setting) Once I disable auto center, this goes away.
Also, Kalman filter seems to work better than Accela.

sthalik, 2020-01-30:
You don't need to turn off auto center. It's enough to center once after the video image has stabilized.

I assume that the first few frames aren't fully exposed. There's nothing special to do about it.

closed 2020-01-30 by sthalik