opentrack/opentrack #1040: No output

Kivasanttu, 2020-01-30:
So im using steamvr as input the pink character on right moves as i turn my mixedreality  headset but what ever output i use it just dosen't do anything, i think im missing a big step somewhere. using it on gta v says its a unknown game, im really new to opentrack and im trying to get my hmd movement to move the mouse so i look left it goes left... dosen't seem to do anything.
im doing bootleg vr gta, i got bigscreen to have monitor infornt of face im using reshade shaders to fake 3d effect
and so far its going good only thing that dosen't work is to look around and the game looks around
i don't have the money for vorpx so im doing it my self whit multiple different softwares works fine but the one thing is missing aka movement

Kivasanttu, 2020-01-30:
tried different games just to get output working but dosen't seem to work

Kivasanttu, 2020-01-30:
Low quality recording, my thing works whit fivem too https://youtu.be/e30ovcquFwM

Kivasanttu, 2020-01-30:
If i get the looking around working i can fake vr whit all the games almost

Kivasanttu, 2020-01-30:
I have tried all of the output options, opening before and after the game, running as adminator thing both game and opentrack, and default both reinstalled, restarted pc, just no output. I think im missing something

sthalik, 2020-01-31:
What is the problem exactly?

Kivasanttu, 2020-01-31:
When i rotate my headset the pink octopus moves, but in game it does nothing

Kivasanttu, 2020-01-31:
i guess im missing another program, or something to use opentrack in games?

Kivasanttu, 2020-01-31:
or am i missing something to do ingame to get it working?

Kivasanttu, 2020-01-31:
today a newday, i tried again plugged in vr, opened steam vr input as valve steamvr output as mouse emulation, and it worked, i don't know what i did wrong last time same thing as before now it worked :/

closed 2020-01-31 by Kivasanttu