opentrack/opentrack #1042: Support protocol selection for Wine

jp7677, 2020-02-02:


sthalik, 2020-02-03:
> Reasoning behind this is that at least DCS via Wine/Proton seems to support both protocols but prefers freetrack when both are available

Did you get DCS to work on Wine?

closed 2020-02-03 by sthalik

jp7677, 2020-02-04:
Thanks a lot for your very fast response! Your implementation seems to works, though I'm a bit confused how the three games I have respond:
Elite Dangerous: Only works with Freetrack (I would expect both to work)
DSC: Works with both, though requires the `headtracker.dl`l in bot cases (I would expect that `headtracker.dll` is only needed for freetrack)
ACC: Only works with NPClient (expected)

> Did you get DCS to work on Wine?

Yes, with SteamPlay/Proton in my case. Only thing needed was vcrun2017. In my case I could install it `withprotontricks 223750 vcrun2017`. Furthermore the F10 key crashes the game. I haven't tried wine without Steam.

sthalik, 2020-02-05:
Did you launch opentrack prior to starting the game? The registry change takes effect only when tracking is started.

jp7677, 2020-02-05:
Yes, I've also checked the registry settings, they were applied correctly. So dunno what else especially ED needs for NPClient. Would be interesting to know what happens on Windows.

Well, anyway, thanks for your work!

commit 50b9638d (2020-02-03): proto/wine: allow disabling freetrack/npclient

Fixes #1042